motioneye-project/motioneye #2173: custom Path not working - save to the path

caslor, 2021-08-14:
Hi to all, i have one small problem and i want your help,
My problem is that my setup cant save the media files are created (movies or pictures) to my external ssd usb HDD

My setup : 
Raspberry pi 4  2GigaRam
Raspberry pi Os lite  - Kernel version: 5.10
Docker : 20.10.8
Portainer : 1.24.2
MotionEye Version : 0.42.1
Motion Version : 4.2.2
OS Version : Debian 9.12
`log2ram` : installed (to extend Sd card life  , `swap` disabled ) 
External usb SSD HDD : Intenso 256 Giga  Fat32

i installed  the  ` usbmount `  to auto mount my external usb drive/Hdd
My mounting point is :` /mnt/usb1`

My setup to the camera is : 
File Storage, Custom Path : `/mnt/usb1/motioneye/Camera1`  (default path was : /var/lib/motioneye/Camera1)

in : `/etc/motioneye/motioneye.conf `   i changed also the path (it was the last thing i did when i saw that everything else didnt work)

```
# default output path for media files (must be writable by motionEye)
#media_path /var/lib/motioneye
media_path /mnt/usb1/motioneye
```

with : `lsblk -fp`
the result is : 
```
NAME FSTYPE LABEL UUID                                 FSAVAIL FSUSE% MOUNTPOINT
/dev/sda
|                                                                     
`-/dev/sda2
     vfat   SSD   0B44-D05F                             238.4G     0% /mnt/usb1
/dev/mmcblk0
|            
```

with :` sudo blkid`
the result is : 
`/dev/sda2: LABEL_FATBOOT="SSD" LABEL="SSD" UUID="0B44-D05F" TYPE="vfat" PTTYPE="atari" PARTUUID="17d64133-02"`

From terminal i can see my mounting point and can explore / create folders  with all my users (root,pi,myuser)
`cd /mnt/usb1`

But i cant find a folder named  "motioneye"  . 
is like the motioneye cant write to the SSD  or cant see it ?

starbasessd, 2021-08-14:
In you r docker run, in addition to:

caslor, 2021-08-14:
Thanks for the Help..

1) Begging from the Fat32 format of my SSD . Thanks for all the interesting info but i will keep it for now in that format as i have seen that the maximum file size (video recording for 60 minutes ) is at about 900mb 

2) About booting from usb , that was my goal and that is why i bought that SSD drive. I have made my raspberry pi 4 board to boot from USB and if doesnt find one to continue in SD card boot.
The problem is that when i try that my board didnt boot from that SSD..   the Green light on raspeberry pi board was always flashing. i dont know if for boot the usb drive needed an external power but as i didnt have any spare i decide to leave that for the future

3) i had all ready modified the fstab  to automount my usb drive 

**4)** To the Original problem now :
I am very new at "Docker" thing  so excuse me if i have questions for obvious things...
After little research based on your answer and the sample of code i manage to run this line of code : 
`
sudo docker run -p  5000:8765 -v /mnt/usb1/motioneye:/mnt/usb1/motioneye --motioneye02 ccrisan/motioneye:master-armhf`
(-p with new port as my original motioneye container used the 8765:8765,  and name : motioneye02  for the new container)

after that i was creat a new container with the name motioneye02  that had running motioneye
and the usb SSD is working fine.

is an other way to add the mounting point  to the original first  motioneye container and didnt understand it?

starbasessd, 2021-08-14:
I usually just completely delete the original container and run the docker as in the Wiki here:
docker run --name="motioneye" \\
    -p 8765:8765 \\
    --hostname="motioneye" \\
    -v /etc/localtime:/etc/localtime:ro \\
    -v /etc/motioneye:/etc/motioneye \\
    -v /var/lib/motioneye:/var/lib/motioneye \\
    -v /mnt/usb1/motioneye:/mnt/usb1/motioneye \\
    --restart="always" \\
    --detach=true \\
    ccrisan/motioneye:master-armhf

You can delete the original docker with a number of commands:
https://www.digitalocean.com/community/tutorials/how-to-remove-docker-images-containers-and-volumes

As to booting from a USB drive, you need to make sure you are on the latest rpi-eeprom update. You can get instructions here:
https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#rpi-eeprom-update
The blinking green light (4 very rapid flashes, followed by a pause, then repeats) usually indicates there's no micro SDCard installed, which is normal.
There is a setting to extend the time allowed to 'discover' the USB drives which corrects a lot of issued, too, within the rpi-eeprom config...

caslor, 2021-08-14:
thanks!!!!!!
So i have to install a new container... ok i understand

as i am using  portainer  , managing  images/containers is very easy through graphical interface 
so  delete the old container is just one click 

thanks about the info for the "extended time for usb"   it is the only thing i havent try about boot from usb (i had check for rpi-eeprom update)

again thanks for the help

starbasessd, 2021-08-14:
BTW, if you go through the issues in github for motioneye, motioneye and portainer don't play well together. You can get it to work, but I've found it's mush more stable without another layer in there. Just my opinion, YMMV