motioneye-project/motioneye #2175: Amcrest ASH26-W (Smart Home) Unable to Open Video Device, on both Home Assistant and Docker

doublepedaldylan, 2021-08-17:
I have been unable to add an ASH26-W camera to motioneye addon for Home Assistant. The RTSP stream is accessible via VLC provided device login credentials.

I have the same error on a separate stock PiOS running docker. 
`ERROR: mjpg client timed out receiving data for camera 2 on port 8082`

On both instances, I have an amcrest IP2M-841 that shows the same errors unless I enter the IP2M's web interface and add a new surveillance user, as I am unable to get a stream with the camera's admin user. The ASH26-W does not have a web interface or any way to add additional device users.

Docker run command used for testing of the issue on the PI:
```
docker run --name="motioneye" \
    -p 8765:8765 \
    -p 8081:8081 \
    -p 8082:8082 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v motioneye:/etc/motioneye \
    -v motioneye_lib:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-armhf
```

starbasessd, 2021-08-17:
What is the url you use to connect to the ASH26-W with VLC, and what are you attempting to use with motionEye in docker?

starbasessd, 2021-08-17:
Who is the manufacturer of the ASH26-W, I was unable to find it in the database on iSpy looking for ASH.

doublepedaldylan, 2021-08-17:
Both are Amcrest cameras.
IP2M (working on motioneye) is accessed with rtsp://192.168.1.XXX/ on both VLC and Motioneye. (url's with channel and substream did not work for me, adding the new user on the device is what finally worked)

ASH26 (not working on motioneye) is accessed with rtsp://192.168.1.YYY/ on VLC. I have tried various channel/subtype url endings, but none of them helped for either case.

starbasessd, 2021-08-17:
Did you add the admin user and password in the appropriate boxes when adding the camera to motioneye?
I would start with: (with admin and password in the appropriate boxes)
rtsp://\[ip_address\]:554
rtsp://\[ip_address\]:554/cam/realmonitor?channel=1&subtype=00&authbasic=YWRtaW46YWRtaW4=\[password\]
rtsp://\[ip_address\]/cam/realmonitor?channel=1&subtype=00&authbasic=YWRtaW46YWRtaW4=\[password\]
rtsp://\[ip_address\]:554/cam/realmonitor?channel=1&subtype=00
\- \- \- \- \-
I would also consider asking Amcrest Support if that particular model (ASH26-W) supports rtsp or http output, as many times companies (REOLINK is notorious for this) of supporting a stream in one model XXXX-YY but not supporting the same stream in the model XXXX-YY-Z.

doublepedaldylan, 2021-08-17:
I did enter username and password in associated boxes in motioneye.
I've tried those variations now with the same results, as well as the link generated from https://www.ispyconnect.com/camera/amcrest (dropping the "username:password@" part of the link, which also did not work.

RTSP stream is accessible on VLC, so it seems the ASH26-W should support stream output, but I can try to check. An Amazon reviewer claimed they got the camera working on Motioneye but gave no details about their config.

starbasessd, 2021-08-17:
VLC is much more forgiving as to what is acceptable in the stream (it will play virtually anything) but motion & ffmpeg are not quite as forgiving....
If you are in a testing mood, you could do a clean install of RaspberryPiOS and either docker-ce or docker.io (I regularly test both) and the instructions for motioneye docker install here (without HA or any other wrappers like portainer):
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
or even the stand alone install:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
and see if you can get it to work there.
I use mostly FOSCAM and D-Link cameras (with the occasional PiCam on the PiZeroW and USB Webcam) without issues in all modes... Pi2B, 3B, 3B+, 4B-4GB, 4B-8GB, and almost any flavor of Linux on Ryzen and Intel hardware, with and without docker...

doublepedaldylan, 2021-08-18:
Minor update: If I find time to dust off my generation 1 Pi B+ I can give a shot at a clean install, however I threw together a zoneminder install on docker, and was able to stream and record the ASH26 with that.

I'd prefer to stay with motioneye, as it's better integrated with Home Assistant, but this may be the route I need to go.

JamBalaya56562, 2026-11-08:
Closing, as the ASH26-W stream could not be opened by Motion both in the add-on and in plain Docker, while VLC and ZoneMinder played it. motionEye passes the RTSP URL to Motion, which reads it with FFmpeg, so a stream it cannot open is for Motion (Motion-Project/motion). Please open a new issue with `motion.log` if a current motionEye and Motion still cannot open it.

closed 2026-11-08 by JamBalaya56562