motioneye-project/motioneye #2176: Movie didn't upload to gdrive

yun00001, 2021-08-18:
Hi I have set my maximum movie length at : 43200 (12hr)/ movie. 
And also set the Auto movie upload to gdrive, but after the movie saved in 2am local, it didn't upload till now. (1pm) GMT+8
Could anyone help me to solve the issue? or is there a limit to upload the video into the gdrive? Thanks.

Spec :  Pi zero W

starbasessd, 2021-08-18:
motionEye or motionEyeOS?
Which version of motion, motionEye, and Base OS? (WebGUI, Settings)
Have you tried lowering the length to 60 minutes, first, then extend to longer versions?
What error in the logs?

yun00001, 2021-08-18:
MotionEye ver : 0.42.1
motionEyeOS : motionEyeOS 20200606
motionOS : 4.2.2+gitUNKNOWN
I havent tried to lower the length to 60 minutes tho, I'll try after this reply

im not sure if its this, but in motioneye.log, it showed up this

2021-08-18 02:18:53: [motioneye]    ERROR: gdrive: file "/data/2021-08-17/14-18-32.mp4" is too large (9741MB/1024MB)
2021-08-18 02:18:53: [motioneye]    ERROR: failed to upload file "/data/2021-08-17/14-18-32.mp4" with service gdrive: file "/data/2021-08-17/14-18-32.mp4" is too large (9741MB/1024MB)
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 87, in upload_file
Exception: file "/data/2021-08-17/14-18-32.mp4" is too large (9741MB/1024MB)

I got the hole file for you so you could check if I'm wrong or correct.
[motioneye.log](https://github.com/ccrisan/motioneye/files/7006601/motioneye.log)

starbasessd, 2021-08-18:
gDrive is telling you that the file is too large.

starbasessd, 2021-08-18:
Also, depending on how large your local drive is, you could rapidly run out of space, and cause it to crash.

yun00001, 2021-08-18:
i see, so i need to lower the video time to fit the 1024MB (1G) per upload.
I got 32G on the os so i think its fine to store about 3 days file into it

yun00001, 2021-08-18:
Is there anyway to let the gDrive upload limit go up to 2048MB?
i'd tried the 60minute and in 1280x720, it will go up to 1.2G and the gDrive wont upload the video file into it.

starbasessd, 2021-08-18:
Lower it to 30 minutes, then.

yun00001, 2021-08-18:
sure.

yun00001, 2021-08-18:
update : 
so I lower the video minute to 45 minute and its under the 1024Mb limit (about 600-700Mb per vid)
and it showed up this.

2021-08-18 23:34:14: [motioneye]    ERROR: failed to upload file "/data/2021-08-18/22-48-54.mp4" with service gdrive: 

is it because the Authorization Key problem? or just reboot so it will retry it again

starbasessd, 2021-08-18:
Please try lowering the file size to 30 minutes. Once you get a working file size, you can start increasing it higher.

yun00001, 2021-08-18:
Okay I'm sorry.

yun00001, 2021-08-18:
2021-08-19 03:12:31: [motioneye]    ERROR: 500 GET /movie/1/download/2021-08-19/02-41-46.mp4?_username=admin&_signature=1c6793faba5d302926cf3eb30af23f9720b817eb (192.168.68.108) 323245.55ms
2021-08-19 03:12:32: [motioneye]    ERROR: 500 GET /movie/1/download/2021-08-19/02-41-46.mp4?_username=admin&_signature=1c6793faba5d302926cf3eb30af23f9720b817eb (192.168.68.108) 308493.57ms
2021-08-19 03:12:43: [motioneye]    ERROR: failed to upload file "/data/2021-08-19/02-41-46.mp4" with service gdrive: 

still got same problem after changing to 30 minute

starbasessd, 2021-08-19:
I am not having issues with 15 minute videos to my google drive on a PiZeroW. 
At this point, I would start over, clean.
Clean image
Set recording time to 15 minutes (to start)
Create a new key for gdrive
and test.

yun00001, 2021-08-19:
Okay i'll try it again.