motioneye-project/motioneye #2178: Failed to create movie

STUKguy, 2021-08-21:
Hi,
I have motion eye installed on docker which i thought was running fine, until i noticed this line that keep reappearing

` ERROR: failed to create movie preview for /var/lib/motioneye/Front: Command '[u'ffmpeg', u'-i', u'/var/lib/motioneye/Front', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/var/lib/motioneye/Front.thumb']' returned non-zero exit status 1,
   ERROR: failed to create movie preview for /var/lib/motioneye/Front: Command '[u'ffmpeg', u'-i', u'/var/lib/motioneye/Front', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/var/lib/motioneye/Front.thumb']' returned non-zero exit status 1`

Anyone know how i can fix this?

Thanks,

starbasessd, 2021-08-21:
Permissions on /var/lib/motioneye & sub-directories?
Do you have a -v /var/lib/motioneye:/var/lib/motioneye in your docker run?

starbasessd, 2021-08-21:
Running portainer or HA as part of the setup?

STUKguy, 2021-08-21:
Motioneye has root permission on docker, this is inside of Portainer. I have 2 other folders which don't have this error. only difference between the camera is Front (issues) is a USB camera and the other 2 are RTSP feeds. Not sure if that would make a differance?

starbasessd, 2021-08-21:
Do you have a -v /var/lib/motioneye:/var/lib/motioneye in your docker run? (portainer setup?)
Does the folder /var/lib/motioneye/Front exist?
Have you tried not using portainer, and use the docker run command from the wiki?

STUKguy, 2021-08-21:
Do you have a -v /var/lib/motioneye:/var/lib/motioneye in your docker run? (portainer setup?)
  - Not sure
  - 
[image]
 
Does the folder /var/lib/motioneye/Front exist?
 - Yes
Have you tried not using portainer, and use the docker run command from the wiki?
 - No

starbasessd, 2021-08-21:
You do have a /var/lib/motion/motioneye but it points to something weird, which again makes me wonder about permissions and existence:
What are the permissions of /srv/dev-disk-by-label-MyCloud/Crucible-CCTV? and does Front exist there?
Are they (all of them) set to "Bind" or "Volume"? THey need to be "Volume"s
If changing them to volumes doesn't fix it, please, please try running the docker run from the wiki.

STUKguy, 2021-08-29:
Sorry its been so late to reply /srv/dev-disk-by-label-MyCloud/Crucible-CCTV has root permissions and "Front" exist there cant remember if they were Bind or Volume. I have had to rebuild my OMV/Portainer server so i am rebuilding the motion eye docker i will use the docker run from wiki, is it this one
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker

STUKguy, 2021-08-29:
You don't happen to know where i would add the USB camera in the container for motion eye would you? for the life of me i cant remember, i think i had to set up an argument or something

starbasessd, 2021-08-29:
Yes to docker wiki.
To forward a video device of your host use an additional parameter like the following

    --device=/dev/video0

STUKguy, 2021-08-29:
Thank you everything is back just need to set it up with SWAG, i will leave this question open for a few days just in case i get that error.

starbasessd, 2021-08-29:
SWAG?

STUKguy, 2021-08-29:
SWAG is a docker container that allows me to set up secure access to other containers such as Motioneye, Plex etc using NGIX

starbasessd, 2021-08-29:
OK

STUKguy, 2021-08-30:
Nope the error is still there
[image]
(changed front to road)

i followed the Wiki guide which i'm sure i used the other time around, i  did notice on the Wiki it doesn't set the PUID & PGID which i don't know if it make a difference but i had to set these up on my other dockers.

starbasessd, 2021-08-30:
I am not able to reproduce, using Raspbian (05/07/2021 fully updated) or Debian 10 (fully updated) and either docker-ce or docker.io installs, running the docker run command from the WiKi. I am not an expert on portainer or SWAG, nor HA, and cannot tell you where to look, other than to point you at their support groups and ask there. As to PUID/PGID it would be for permissions to folders/files/paths. Since motionEye assumes you are running it as a user (root/admin) that has permissions to the paths, it isn't specified. When you throw other layers into the mix, (running a docker container as other than the expected user, with no permissions to the folders/files/paths) then you can run into that kind of issue. There are permission issues when running motionEye as a 'normal' user, and documentation in the WiKi. For your portainer/SWAG, I'll leave it to you or others, as I do not see a large contingent of users with the problem for me to build such an environment at this time. Here is the WiKi link:
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User 
If you want to let me know where your install instructions for SWAG & NGINX, if I get time, I will consider the issue further.

STUKguy, 2021-08-30:
I'm not asking you to help with SWAG or HA i haven't even installed it yet, i'm asking for help with the issue which i put in the first post (apologise for mentioning SWAG as it seems to have confused you)

`ERROR: failed to create movie preview for /var/lib/motioneye/Front: Command '[u'ffmpeg', u'-i', u'/var/lib/motioneye/Front', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/var/lib/motioneye/Front.thumb']' returned non-zero exit status 1, ERROR: failed to create movie preview for /var/lib/motioneye/Front: Command '[u'ffmpeg', u'-i', u'/var/lib/motioneye/Front', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/var/lib/motioneye/Front.thumb']' returned non-zero exit status 1`

which has come back even after rebuilding the MotionEye docker, i have checked all permission and they are all root. MotionEye creates the file in the correct place the only issue that i seem to have is the there is no image preview i just seem to end up with this
[image]

I mentioned to PUID & PGID in case you thought adding that might help fix this problem?

starbasessd, 2021-08-30:
How are you starting the motionEye docker? 
What user invokes the command?

STUKguy, 2021-08-30:
It starts automatically, i don't have to touch anything for this issue to come up. i have restarted to the motion eye docker now and it will be fine for a few hours and then the error starts to come up

STUKguy, 2021-08-30:
i haven't put any scripts on boot but Portainer automatically starts with the server (OMV5)  which in turn starts all my docker containers.

starbasessd, 2021-08-30:
Can you stop it in portainer, and run the stand alone docker run command to test?
Did you change the BIND connections to VOLUME connections in portainer?

STUKguy, 2021-08-30:
What is the "stand alone docker run command to test"
Yes that was the first thing i did this time around

starbasessd, 2021-08-30:
The docker run command found here:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker
If running on a pi change the last line to armhf instead of amd64
Change the volume locations as appropriate for your setup.

STUKguy, 2021-08-30:
Ah ok sorry i thought there was another "run" command i was missing but no i have ran this one just changed the volumes.

starbasessd, 2021-08-30:
Let me know.

STUKguy, 2021-08-31:
I'm not having the best of luck with my server, woke up this morning to find my SSD had died :/ anyhow i am trying to setup motion again but for some reason when i try and add the USB camera i keep getting this error
[image]

this is the line i am adding
[image]

and these are this is what comes up when i check the camera names on my server
[image]

i did everything the same as i always have but for the life of me i cant figure out why its not working now, do you have any ideas?

starbasessd, 2021-08-31:
What are you using, specifically? Portainer or something else? You want to 'pass through' the video device. The line you put in there is for a direct docker run command.
Try removing the '--device=' from the second box

closed 2021-08-31 by STUKguy

starbasessd, 2021-08-31:
Did that fix it?