motioneye-project/motioneye #2179: H264

Oxize, 2021-08-24:
When can we see an MotionEye version with H264 compression, instead MPJEG. Nowadays (all my cams are can handle 5G) most cams can handle much higher bandwith, which also readyh for H264.

We add now for example 3 camera ip, you need to use all MPJEG for those. 

With comming of PI4, which have 1GB connection on Ethernet and 5G on wifi. Enough to stream H264. Looks now MotionEye is on a stillstand. 

Or did something happen to the owner?

starbasessd, 2021-08-24:
h264-omx is available, as is rtsp, both input and output.
Pi 4 still has limitation of about 800Mb/sec.
Pi 4 does NOT support 802.11ax, it does do 802.11ac but maxes out at about 85Mb/sec.
The owner can answer for himself.

closed 2021-08-24 by starbasessd

Oxize, 2021-08-24:
Ezviz Studio here on my pc has 3 cams on it. They all run in H264, and going through an 5ghz Unifi AP. No stuttering. I use MotionEye for to check stuff remote. Oh well lets see what the future brings.

starbasessd, 2021-08-24:
MotionEyeOS on the Pi4 (dev20201026) will give the 'best performance' among the Pi SBCs especially if overclocked to 2.0 or 2.1 GHz, but compared to a PC with 11.ax won't even come close. The non-Intel SBCs are overloaded for performance compared to Intel or AMD CPUs. 11.ax is Gb/s capable, 11ac (especially on the limited wifi chip on the Pis) can't come close.