motioneye-project/motioneye #2180: Raspbian bullseye support?

Gonioul, 2021-08-24:
Hello,

I suppose it's linked to python3 support, but just in case, I'm opened to feedback.

Regards.

starbasessd, 2021-08-24:
Yes, Debian 11 is not supporting Python2.7, required for motionEye, at this time.

chey, 2021-11-08:
It would be better if motionEye supported Python3. But if you run into the situation where you installed bullseye on your Pi and want to setup motionEye you can get `pip` for Pythong2.7 manually:

1. First install python2.7 `apt-get install python2.7`.
2. Then follow the instructions [here](https://github.com/pypa/get-pip) (replace `python` command with `python2.7`:

After following the **Usage** instructions at the link that should get you `pip2` and you can continue with the rest of the motionEye installation instructions using `pip2`.

starbasessd, 2021-11-08:
Instructions for Debian Bullseye support updated, tested, working.

closed 2021-11-08 by starbasessd

starbasessd, 2021-11-08:
Instructions for RaspberryPiOS Bullseye updated, tested, confirmed working.