motioneye-project/motioneye #2182: Adding RTSP stream results in grey preview, errors in log

voidnecron, 2021-09-02:
Hi,

I have a working MotionEyeOS on RPi 3b and a second MotionEye install on Rasbian 10. With both versions the attached USB cam (a PS3 Eye) works fine, but I can't get the RTSP added correctly. I think it has something to do with the characters in the URL which might not be escaped properly and therefor part of the URL gets dropped.
Some info:
# uname -a
Linux domoticz 5.4.61-v7+ #1339 SMP Tue Sep 1 18:45:30 BST 2020 armv7l GNU/Linux
# cat /etc/debian_version 
10.10
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Raspbian 10
Camera is http://www.escam.cn/product/66-en.html
RTSP stream URL is syntax as: rtsp://10.6.3.57:554/user=admin&password=&channel=1&stream=0.sdp?

I tried to add the camera as:
rtsp://10.6.3.57/user=admin&password=&channel=1&stream=0.sdp? (blank field for user/pass in GUI)
rtsp://10.6.3.57:554/user=admin&password=&channel=1&stream=0.sdp? (blank field for user/pass in GUI)
rtsp://10.6.3.57:554/channel=1&stream=0.sdp? (filled fields for user/pass in GUI)
rtsp://10.6.3.57:554/channel=1&stream=0.sdp (filled fields for user/pass in GUI)
and all the above with single and double quotes, but the GUI doesn't accept quotes so you can't add the stream that way.

Does anyone have a clue why this could be happening? VLC plays the stream correct, but I've already read that VLC is very forgiving and motion(eye) and ffmpeg aren't...

Cheers!

starbasessd, 2021-09-02:
Not usually. 
There are literally dozens of entries for the QD500 at:
https://www.ispyconnect.com/camera/escam
with fixed admin/password combinations, as well as variable. It may be that your specific camera has one of those? If your password is user-setable, could you have a non-valid character in the password? (like a '&') in it?

voidnecron, 2021-09-02:
> Not usually.
> There are literally dozens of entries for the QD500 at:
> https://www.ispyconnect.com/camera/escam
> with fixed admin/password combinations, as well as variable. It may be that your [...]

To be sure of all variables I completely removed the password, so it's just the username 'admin' with no password at the moment. But as I mentioned and you now as well, could it be the & (ampersand sign) in the URL?

starbasessd, 2021-09-02:
& is a special trigger character (as is ? and a few others) and is used quite often in other camera URLs without issue.

starbasessd, 2021-09-02:
When adding a camera, Add Camera, Network, you get the box:
[image]
For the URL, you use (for example)
rtsp://10.6.3.57:554/channel=1&stream=0.sdp?
You use Username: admin
You use Password: \[whatever\]
For Camera: Have you selected for both the TCP and UDP versions?
Once the camera is added, do you select the native resolution of the camera, or something else? I can get gray screen or green screen if I select the wrong resolution. My Foscams want me to select a 16x9 (I use 1280x720 normally)

zagrim, 2021-09-04:
Many of those URLs look very suspicious, I mean why would the manufacturer have build a custom URL query parameter parser to pick then up in the path (the format with parameters coming after slash (/)) when probably any HTTP server has the standard URL parsing built-in. So, what if you tried
```?user=admin&password=tlJwpbo6&channel=1&stream=0.sdp```
instead of 
```/user=admin&password=tlJwpbo6&channel=1&stream=0.sdp```
or
```/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp```
(So, put question mark at the front or the parameters and separate them with ampersands, and ditch any remaining question marks from the examples)

As an example with full URL:
```rtsp://10.6.3.57:554?channel=1&stream=0.sdp```

JamBalaya56562, 2026-11-08:
Closing, as this got no answer to zagrim's suggestion and was with motionEye 0.42.1 and Motion 4.2.2. This camera puts its login in the path (`/user=admin&password=...&channel=1&stream=0.sdp`), so it has to be in the URL itself, with the Username and Password fields left empty, or as a query string (`/?user=admin&password=...`) as zagrim suggested. Please open a new issue with `motion.log` if such a URL still does not work with a current release.

closed 2026-11-08 by JamBalaya56562