motioneye-project/motioneye #2183: motioneye service fails to start properly after reboot with mjpg

Gonioul, 2021-09-02:
Hi,

I'm using motioneye on a rasbian pi4 as a frontend for pi zero cameras on fast camera mode.

Using systemd service file provided in the doc, when rebooting the main raspberry I have:
"ERROR: mjpg client timed out receiving data for camera 1 on port 8081"
and no videos.

Works fine after a manual ssh "systemctl restart motioneye"

After searching, for /usr/local/share/motioneye/extra/motioneye.systemd-unit-local and then /etc/systemd/system/motioneye.service :
[Unit] needs: After=network-online.target

[Install] would probably need: After=network-online.target

Strange mjpg is not retrying properly when network fails or not ready...

starbasessd, 2021-09-02:
Is RaspberryPiOS fully updated?
WiFi or Ethernet?
Have you turned on 'Wait for Network'?
Static IP or DHCP?

Gonioul, 2021-09-02:
No problem now, After=network-online.target seems to fix the problem.

JamBalaya56562, 2026-11-08:
Closing, as this no longer happens. On a Raspberry Pi 5 on Wi-Fi with motionEye 0.45.0a1, Motion 4.7.1 and the unit file as shipped (`After=network.target`), motionEye and Motion still start before the Pi has its address (at 7.2 s and 9.9 s, against 10.3 s), and a network camera at the Pi's LAN address fails its first image, but Motion then connects to it by itself about 30 s after boot, in two reboots, with no restart needed. If you prefer motionEye to wait for the network anyway, a local override with `Wants=network-online.target` and `After=network-online.target` does that.

closed 2026-11-08 by JamBalaya56562