motioneye-project/motioneye #2184: Now I just want to start Motioneye manually

RPiDespair, 2021-09-05:
Hi,

motionEye Version 0.42
Motion Version 4.2.2
OS Version Raspbian 10

MotionEye starts automatically when the system starts.
MotionEye running as normal user.
Now I just want to start Motioneye manually.
What is to be changed?

Regards
RPiDespair

starbasessd, 2021-09-05:
To stop motioneye:
sudo systemctl motioneye stop
To start motioneye:
sudo systemctl motioneye start

starbasessd, 2021-09-05:
Also: 
at the end of 'Run as a Normal User':
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User

Manually Started motionEye Server

Assuming sudo is available on your system, the command should look something like this:

sudo -u youruser meyectl startserver -b -c /home/youruser/motioneye/motioneye.conf

starbasessd, 2021-09-05:
Sorry, my bad:
To stop motioneye:
sudo systemctl stop motioneye
To start motioneye:
sudo systemctl start motioneye

RPiDespair, 2021-09-06:
Hi starbasessd,

thank you very much for your help.

> To stop motioneye:
> sudo systemctl stop motioneye
> To start motioneye:
> sudo systemctl start motioneye
I already knew that.

`sudo -u **RPiDespair** meyectl startserver -b -c /home/**RpiDespair**/motioneye/motioneye.conf`
`sudo reboot`

Motioneye starts again automatically.
Unfortunately it didn't work.

starbasessd, 2021-09-06:
sudo rm -f /etc/systemd/system/motioneye.service
This removes the ability to use systemctl to start or stop the service (and no longer will start at boot)
start
 sudo -u RPiDespair meyectl startserver -b -c /home/RPiDespair/motioneye/motioneye.conf
stop
 sudo -u RPiDespair meyectl stopserver -b -c /home/RPiDespair/motioneye/motioneye.conf

RPiDespair, 2021-09-06:
Hi starbasessd,
Thank you, that's exactly what I was looking for.

closed 2021-09-06 by RPiDespair