motioneye-project/motioneye #2185: Suggestion for daemon

mydomdom, 2021-09-06:
It would be nice for newbies that are on a raspberry
-  either to set 'start_motion_daemon=yes' in '/etc/default/motion' instead of 'no', 
- or to state that a change should be made to this file in step 7 of [https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian].

That would save some time to people like me searching why motioneye isn't loaded at startup.
Fortunately electron1979 gave the solution in [https://github.com/ccrisan/motioneye/issues/900].

Impressed by this work guys !

starbasessd, 2021-09-06:
The instructions for running as a daemon are pretty clear.
The instructions for installing and running at boot are here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
or Running as a non-root user are here:
https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User
What are you trying to accomplish?

Gonioul, 2021-09-06:
Motioneye is managing it's own instance of motion, so start_motion_daemon should be set to no in /etc/default/motion
If you set start_motion_daemon to yes, you'll have another instance of motion (not motioneye) with default config, that is trying to access /dev/video0 or something similar and loop failing or locking the device.
That's not clear in the install doc.

starbasessd, 2021-09-06:
Where is motion getting started?
What set of install instructions are you using?
Following the normal install instructions found in the motionEye github wiki, motion daemon is not getting started by itself at boot on any of my test machines (RPi, VBox VMs running Ubuntu, Debian, etc)

Gonioul, 2021-09-06:
I agree with mydomdom, motion package defaults start_motion_daemon to no, but motioneye guide doesn't say a word about it.
People would tend to set start_motion_daemon to yes if it doesn't work, so a line in the doc would be nice to prevent it.

starbasessd, 2021-09-06:
"motion package defaults start_motion_daemon to no" and 'newbies' making random changes is not something that can be prevented by a line in a document.

Gonioul, 2021-09-06:
Well, nothing in the wiki says motioneye is managing it's own instance of motion and not managing the system one...

JamBalaya56562, 2026-11-08:
Closing, as this is done since motionEye 0.43.1: `motioneye_init`, the install step of the README, now stops and disables the system's `motion.service` when it runs or is enabled (#2980), since motionEye starts its own Motion. So `start_motion_daemon` in `/etc/default/motion` no longer needs any attention.

closed 2026-11-08 by JamBalaya56562