motioneye-project/motioneye #2186: Bad gateway

fabry891, 2021-09-07:
Hi, I've installed home assistant and motioneye as docker. Home assistant work very fine.
But if I try to open the motioneye web interface, It show "502: bad gateway". I try with different browser.
What is the cause?

This is motioneye log:
```
[00:27:49] INFO: Starting motionEye...
[00:27:49] TRACE: bashio::net.wait_for 28765
[00:27:49] TRACE: bashio::net.wait_for 28765
    INFO: hello! this is motionEye server 0.42.1
   DEBUG: using motion version 4.3.2
   DEBUG: using ffmpeg version 4.4
   DEBUG: reading main config from file /data/motioneye/motion.conf...
   DEBUG: loading additional config structure for main, without separators
   DEBUG: additional config section: network
   DEBUG: listing config dir /data/motioneye...
   DEBUG: listing smb mounts...
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: smb mounts started
    INFO: server started
[00:27:49] INFO: Starting NGinx...
[00:27:49] TRACE: bashio::discovery: motioneye {"url":"http://127.0.0.1:28765"}
[00:27:49] TRACE: bashio::api.supervisor POST /discovery {"service":"motioneye","config":{"url":"http://127.0.0.1:28765"}} .uuid
[00:27:49] TRACE: bashio::var.has_value: {"service":"motioneye","config":{"url":"http://127.0.0.1:28765"}}
[00:27:49] DEBUG: Requested API resource: http://supervisor/discovery
[00:27:49] DEBUG: Request method: POST
[00:27:49] DEBUG: Request data: {"service":"motioneye","config":{"url":"http://127.0.0.1:28765"}}
[00:27:49] DEBUG: API HTTP Response code: 200
[00:27:49] DEBUG: API Response: {"result": "ok", "data": {"uuid": "43528841d0554288be3020f56fcbe95e"}}
[00:27:49] TRACE: bashio::jq: {"result": "ok", "data": {"uuid": "43528841d0554288be3020f56fcbe95e"}}
 .result
[00:27:49] TRACE: bashio::var.true: {"service":"motioneye","config":{"url":"http://127.0.0.1:28765"}}
[00:27:49] TRACE: bashio::jq: {"result": "ok", "data": {"uuid": "43528841d0554288be3020f56fcbe95e"}}
 if .data == {} then empty else .data end
[00:27:49] TRACE: bashio::var.has_value: .uuid
[00:27:49] DEBUG: Filtering response using: .uuid
[00:27:49] TRACE: bashio::jq: {"uuid":"43528841d0554288be3020f56fcbe95e"} .uuid
[00:27:49] TRACE: bashio::cache.flush_all
[00:27:49] TRACE: bashio::fs.directory_exists: /tmp/.bashio
[00:27:49] INFO: Successfully send discovery information to Home Assistant.
   DEBUG: running cleanup process...
   DEBUG: cleaning up pictures...
   DEBUG: cleaning up movies...
   DEBUG: cleanup done
   DEBUG: checking SMB mounts...
   DEBUG: listing smb mounts...
   DEBUG: checking SMB mounts...
   DEBUG: listing smb mounts...
   DEBUG: checking SMB mounts...
   DEBUG: listing smb mounts...
   DEBUG: checking SMB mounts...
   DEBUG: listing smb mounts...
```

starbasessd, 2021-09-08:
Which docker run command are you using for motioneye?
motionEye docker uses the same port as motioneye, 8765 for the WebGUI. 
For some reason it is mentioning 28765.

fabry891, 2021-09-08:
Hi, thanks for your reply.
I don't know which docker run command I'm using for motioneye. I've installed it via home assistant - supervisor page.
Do you suggest install motioneye via docker run command?

starbasessd, 2021-09-08:
I recommend not using HA or Portainer to supervise motionEye in a docker container. 
I have several systems running docker in Ubuntu Server, Debian Server RaspberryPiOS Lite, in VMs and RPi3B+s and RP4s, and all run well as their own docker. There are many tickets open (and closed) where there are issues with HA and Portainer in the motionEye Github and Google groups. When run with it's own docker run, connected to HA, it's easier for us to see any issues.

starbasessd, 2021-09-11:
What have yu found running motionEye docker container by itself?

JamBalaya56562, 2026-11-08:
Closing, as the 502 came from the Home Assistant add-on's proxy in front of motionEye (port 28765 is the add-on's, not motionEye's 8765), and this got no answer about running motionEye on its own. Such issues belong to the add-on's repository; please open a new issue here if motionEye itself does not answer on its port with a current release.

closed 2026-11-08 by JamBalaya56562