motioneye-project/motioneye #2187: Normal user timelapse not working

Gonioul, 2021-09-13:
Timelapse defaults to omx codec which need video group.

/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py

Either remove "_omx" for software codec, or add normal user to video group.

Should be added to wiki: https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User

starbasessd, 2021-09-13:
What board, and what base OS?

Gonioul, 2021-09-13:
rasbian buster, pi4

starbasessd, 2021-09-15:
What time lapse mode?

Gonioul, 2021-09-15:
Any, default will do, as the problem is the codec, not the settings (group, every, framerate).
I suppose the problem shows up only when *_omx codecs are available to unpriv user.
Raspbian seems to limit those codecs to video group, maybe for hardware acceleration access.

JamBalaya56562, 2026-11-08:
Closing, as the timelapse is not tied to omx: it is encoded with the camera's Movies > Movie format, in 0.42.1 as today, so it uses `h264_omx` only when that format is chosen for the movies too. A user running motionEye needs to be in the `video` group to use the Pi's hardware encoders, or the camera can use a software format such as H.264 (.mp4), which works for any user.

closed 2026-11-08 by JamBalaya56562