motioneye-project/motioneye #2188: Documentation: Add mention that motion server should not be started automatically

emeitner, 2021-09-15:
The documentation should add a note that the motion server, if installed as a package, should be set to not automatically start. Not doing so will produce the dreaded "ERROR: mjpg client timed out receiving data for camera 1 on port
 8081" in the log and prevent the camera from working.
In Debian this can be done with:
` systemctl disable motion.service`

starbasessd, 2021-09-15:
Considering motion is not set to 'start automatically' in any of the instruction sets in the Wiki:
https://github.com/ccrisan/motioneye/wiki/Installation
Where was it suggested that it be enabled?

emeitner, 2021-09-16:
The issue is that installing "motion" under Debian configures systemd to automatically start it. This caused me 1 hour of troubleshooting. A simple note in the Wiki stating that after installing "motion" it should be disabled would suffice.

starbasessd, 2021-09-17:
Using Debian 10 (Buster) fully updated:
su -
apt update && apt upgrade -y
apt-get install motion ffmpeg v4l-utils
systemctl status motion
● motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion; generated)
   Active: active (exited) since Thu 2021-09-16 20:46:42 EDT; 28s ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 4700)
   Memory: 0B
   CGroup: /system.slice/motion.service

Sep 16 20:46:42 deb10-mstr systemd[1]: Starting LSB: Start Motion detection...
Sep 16 20:46:42 deb10-mstr motion[6653]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Sep 16 20:46:42 deb10-mstr systemd[1]: Started LSB: Start Motion detection.
ps auxw | grep motion
root      6907  0.0  0.0   6076   892 pts/0    S+   20:49   0:00 grep motion
sudo systemctl list-unit-files
motion.service                                                            generated
reboot
Same results. Motion is not running in top. Doesn't show in ps auxw. Where is it running in your system?
Testing further:
sudo systemctl stop motion
sudo systemctl disable motion
sudo systemctl daemon-reload
continue with motioneye install
sudo apt-get install python-pip python-dev python-setuptools curl libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev
pip install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local \  
&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye

starbasessd, 2021-09-17:
Added 
systemctl stop motion
systemctl disable motion
after apt-get install motion ffmpeg v4l-utils
in Debian Install Wiki.

closed 2021-09-17 by starbasessd

zagrim, 2021-09-17:
I think I ran into this while updating my desktop (which has Motion installed for development purposes) from Buster to Bookworm (current testing since Bullseye was released). Anyway, there was this in the changelogs:

> motion (4.3.2-1) unstable; urgency=medium
> 
>   Migrate to systemd instead of initscript. This no longer honours
>   /etc/default/motion, so if you had disabled the service previously then you
>   should run `systemctl disable motion` after upgrading to this version.
> 
>  -- Ximin Luo <infinity0@debian.org>  Sun, 15 Nov 2020 20:07:37 +0000

So I think that addition to the instructions was indeed proper. I needed to disable Motion that way, too. I'm just not quite sure if that change is from Bullseye or from Bookworm since by accident, or chance, I skipped over Bullseye (and got the worst upgrade experience of my life due to it - kids, never skip a release when upgrading).