motioneye-project/motioneye #2189: Cannot disable motion detection

matej2, 2021-09-15:
I was trying to disable motion detection (and command triger that is configured). On UI, it seems like it is disabled, but the command still triggers. I have managed to find relevant logs, I will provide more if needed:

```
Sep 15 18:41:07 raspberrypi systemd[1]: Started motionEye Server.
Sep 15 18:41:16 raspberrypi meyectl[268]:     INFO: hello! this is motionEye server 0.42.1
Sep 15 18:41:33 raspberrypi meyectl[268]:     INFO: cleanup started
Sep 15 18:41:33 raspberrypi meyectl[268]:     INFO: wsswitch started
Sep 15 18:41:33 raspberrypi meyectl[268]:     INFO: tasks started
Sep 15 18:41:33 raspberrypi meyectl[268]:     INFO: mjpg client garbage collector started
Sep 15 18:41:34 raspberrypi meyectl[268]:     INFO: server started
Sep 15 18:41:34 raspberrypi meyectl[268]:    ERROR: failed to disable motion detection for camera with id 1: Failed to connect to 127.0.0.1 port 7999: Connection refused
```

I hope I did provide enought information. Thanks for your help

starbasessd, 2021-09-15:
How are you trying to disable motion detection?
WebGUI port 8765 or motion port 7999? If port 7999 what are you using for your command?

matej2, 2021-09-15:
WebGUI. I was editing camera configuration

starbasessd, 2021-09-15:
There are several different ways to 'disable' motion detection, each with different indications and results.
In Still Images, select 'Capture Mode: Manual'
In Motion Detection, Select Title Bar: Off'
Still Images does not disable detection completely.
Motion Detection DOES disable it completely, but only for a particular camera

If the motion web interface is enabled:
{IP}:{port}/{camid}/detection/start Start or resume motion detection.
{IP}:{port}/{camid}/detection/pause Pause the motion detection.
Send command: curl http://ip_address:7999/1/detection/pause
It will not show in the WebGUI.
More Info can be found here:
https://motion-project.github.io/motion_config.html#OptDetail_Webcontrol

matej2, 2021-09-16:
Still images are completelly disabled, motion detection was disabled completelly. I have only one camera.

So I directly connected to device and ran curl to localhost. This worked. However, disabling it using UI does not work, and returns me the same error that I pasted in this issue.

starbasessd, 2021-09-16:
I am confused. You state that you are trying to disable motion detection, but you also state it is already disabled, and yet it is still recording triggered motion?
Can you show Settings, Still Images, Settings, Movies, and Settings, Motion Detection, please? (Screen shots would be fine)

matej2, 2021-09-16:
Yes, exactly. That is exactly what I wrote in issue description. This is the bug that i am trying to report.
[image]
[image]

starbasessd, 2021-09-16:
What do you have set in Settings, Movies?

matej2, 2021-09-16:
[image]

starbasessd, 2021-09-16:
OK, will build and look at it this afternoon.

matej2, 2021-09-19:
I had some problems with OS so I reinstalled it and reproduced the issue. Here are some more details, hoping this would help:

- I installed Raspbian OS lite (no desktop)
- changed port from 8765 to 80 as defined here: https://github.com/ccrisan/motioneye/wiki/Configuration-File#port
- set up Dataplicity
- enabled camera (the one that is connected directly to pi)

I tried different configuration combinations, like enabling everything and then disabling motion detection. The error (Connection refused) still appears.

It appears that this is the relevant error: https://github.com/ccrisan/motioneye/blob/795616d15caa24fd35f1d41b37668086e1ce4049/motioneye/motionctl.py#L265

starbasessd, 2021-09-20:
Before making any other changes, does the WebGUI work on 8765?
You can't open the Motion Detection section with it turned Off. If you can, you've broken it somewhere else.

matej2, 2021-09-21:
I will check.

If you are talking about screenshot I posted earlier: Motion detection was turned off, I expanded it to take screenshot

starbasessd, 2021-09-21:
When set to on on my machines, it opens the section
When set to off on my machines, it closes the section.
I cannot get it to stay open when set to off, in either motionEye on another OS or motionEyeOS.
What you posted for a screenshot should not be possible.

matej2, 2021-09-22:
There are two screenshots

matej2, 2022-05-24:
Solved, I created a screenshot with different app