motioneye-project/motioneye #2191: Streaming Image Resizing

akhilleusuggo, 2021-09-15:
Hello;

I'm new using this awesome software, and I would like to thank you for it.

I'm facing a little problem here with the Streaming Image Resizing option. It doesn't work when watching the stream directly from the TCP port indicated ( default 8081 ).

It does work only when watching the stream from the main UI. ( default port 8765 ).
Is it supposed to be like this, or is there some configuration that I have to change.

Thank you in advance.

starbasessd, 2021-09-15:
It affects the resolution, but CSS affects the screen size... There's a whole discussion in a recent issue thread about it.

zagrim, 2021-09-16:
Yes, please see #2162 for further details. A direct quote from comments there: "[Streaming Image Resizing] is designed to affect only the frames in the camera video stream MotionEye server sends to the user interface".

akhilleusuggo, 2021-09-16:
I see, thank you for the answer. I followed up with the thread, so I would like to ask you, if there's anyway to deny/block to stream via port default TCP port 8081.

The main goal for me was to reduce the bandwidth.

starbasessd, 2021-09-16:
I am confused. 
Turn it off?  Settings, Video Streaming, set title bar to Off.
Reducing bit rate?
Reducing Frame Rate via Settings, Video Streaming, Stream Frame Rate to lower number.
Reducing Streaming Quality via Settings, Video Streaming, Streaming Quality works like reducing jpg quality.

akhilleusuggo, 2021-09-16:
I mean, since "Streaming Image Resizing" only works for user interface, I would like to disable direct access to the stream, and allow only from the user interface.

Also, I'm not sure why, but it doesn't seem I can find where is the FFmpeg lines to tweak. Right before I used motioneye, I used motion ( this is what led me to the motioneye ), and on the configuration file was possible to tweak the preset/bitrates etc. of the stream/movies. Here I'm not able to find it.

starbasessd, 2021-09-16:
in motioneye: /etc/motioneye/motion.conf
in motionEye WebGUI: Settings, Video Device, Extra Motion Options (one per line)
Using the web interface for motion: 
https://motion-project.github.io/motion_config.html#OptDetail_Webcontrol

akhilleusuggo, 2021-09-16:
> in motioneye: /etc/motioneye/motion.conf

```
# @enabled on
# @normal_password
# @admin_username admin
# @admin_password (no idea yet what this password is for)
# @normal_username user

webcontrol_interface 1
webcontrol_port 7999
setup_mode off
webcontrol_parms 2
webcontrol_localhost off

camera camera-1.conf
```

Not sure what do you mean, but I'll read the doc first. Thank you for the quick reply. As for the main issue is solved.
Just disabled the streaming option, and works as expected for me. 

Thanks

zagrim, 2021-09-17:
> I would like to ask you, if there's anyway to deny/block to stream via port default TCP port 8081.

For that specific question there is a specific answer: You could set up firewall rule(s) for that. The details for that depend somewhat on the setup (which base OS/distro etc) but basically that could be done with something like `sudo iptables -A INPUT -i wlan0 -p tcp --dport 8081 -j REJECT` (replace `wlan0` with the network interface through which clients connect to ME). The details how to make that rule persistent over reboots will also depend on your setup.

akhilleusuggo, 2021-09-17:
@zagrim exactly what I did, thank you for the reply.
Also just disabled the stream video tab ( but via the UI interface works, and that was my confusion at the beginning ).

JamBalaya56562, 2026-11-10:
Closing with the answer, as in #2162 the resizing applies only to the UI. To keep the stream from being opened directly, turn Video Streaming off: Motion then serves it on localhost only, and the UI still shows the camera through motionEye.

closed 2026-11-10 by JamBalaya56562