motioneye-project/motioneye #2192: Maximum movie length - misleading behaviour

AHeroicLlama, 2021-09-20:
I have been able to observe (via using MotionEyeOS) that the maximum movie length parameter does _not_ define the maximum length of the outputted movie - instead it is ignorant of time skipped due to 'motion gaps' and it instead cuts the movie as soon as the 'maximum length' has elapsed in _real time_.

I have also observed that in this scenario, another movie _will not_ be created immediately after containg the remaining motion, despited tooltips suggesting so.

Example;
Maximum movie length 60s and motion gap 30s.
A motion event starts at 12:45:00 and lasts for 5 seconds. Then 20 seconds later at 12:45:25 another 5 seconds of motion is recorded. (These 2 lots of 5 seconds are stitched together into one movie lasting 10 seconds.) Then a further 20 seconds later at 12:45:50 a 20-second period of motion starts, but MotionEye will cut the movie _mid-motion_ at 12:46:00 because it has been 60s elapsed since the first motion event.
This results in a movie which should have been 30 seconds of motion being cut to just 20 seconds. This is much below the maximum movie length, and more importantly drops recorded motion.

Obviously it is debatable if this is mis-documented or mis-implemented, but my impression as a user is that it is mis-implemented, as it is more valuable to cap the length of the movie file due to file system/storage limitations. I would suggest it is not useful to cap the motion to a real-time window of time which is ignorant of motion detection.

starbasessd, 2021-09-20:
My interpretation of maximum movie length is to be used when you anticipate any one movie generated will be greater than 'X' amount of time in order to 'break up' individual file size.
Selecting for motion gap of 30 seconds between triggers should mean trigger1 starts at 12:00:00, and motion ends at 12:00:05, the next motion trigger event should not enable until 12:00:35. This may or may not meet your needs/requirement. To have multiple trigger events during the time 12:00:00 to 12:00:30 would cause many race conditions, multiple trigger trackers would need to be used, or reset timer of :30 to :00 each time, which then defeats the purpose of the max length.
It has be requested to limit to space availability, but that is equally hard to implement. What if you have 2 GB available, but then get caught by a long motion, and it still crashed the system because the video is creating a >2GB file? I think it would take a major re-write to create a FIFO continuous loop file system where it starts a file, writes to end, then starts over at the beginning, like a tape loop or 8 track tape...

zagrim, 2021-09-21:
@AHeroicLlama  Well, MotionEye does not do anything other here than to display the tooltip, so if you think the implementation is incorrect of flawed you should address the Motion project directly. However, I don't think the tooltip is suggesting the next movie will be created _immediately_ as you seem to have interpreted it.
Motion docs say next to nothing about [movie_max_time](https://motion-project.github.io/motion_config.html#movie_max_time), but a great deal of [event_gap](https://motion-project.github.io/motion_config.html#event_gap).

Anyway, the way I see it after reading the Motion documentation, if you don't want ever to miss any piece of motion, you would have to have "Capture Before" (in seconds, calculated from the camera frame rate) equal to "Motion Gap". In that case, (I just woke up so I might not be at my brightest yet) the example you gave should work so that there is no gap between the movies. However, Motion docs warn of having large [pre_capture](https://motion-project.github.io/motion_config.html#pre_capture) since it can affect performance greatly (the limits show in docs might be for quite a low-end system, modern desktop/laptop might be able to deal with much longer buffers).

Anyway, please read the linked parts of the Motion docs (and perhaps also the part on "post_capture" that comes right after those") and see if they give you any further insights on how to configure your setup to your requirements.

JamBalaya56562, 2026-11-08:
Closing, as Maximum Movie Length is Motion's `movie_max_time`, which motionEye only passes on, as zagrim explained: how it counts the time and when the next movie starts is up to Motion. If that behaviour should change, the Motion project (Motion-Project/motion) is the place to ask.

closed 2026-11-08 by JamBalaya56562