motioneye-project/motioneye #2193: Broken build of Docker image from dev branch

yurnov, 2021-09-25:
When I reconfigured my installation of motioneye I decided to rebuild the docker image and faced that the build is broken:

```
 > [3/4] RUN echo "deb http://snapshot.debian.org/archive/debian/$(date +%Y%m%d) sid main contrib non-free" >>/etc/apt/sources.list &&     apt-get update &&     DEBIAN_FRONTEND="noninteractive" apt-get -t stable --yes --option Dpkg::Options::="--force-confnew" --no-install-recommends install       curl       ffmpeg       libmicrohttpd12       libpq5       lsb-release       mosquitto-clients       python-jinja2       python-pil       python-pip       python-pip-whl       python-pycurl       python-setuptools       python-tornado       python-tz       python-wheel       v4l-utils &&     DEBIAN_FRONTEND="noninteractive" apt-get -t sid --yes --option Dpkg::Options::="--force-confnew" --no-install-recommends install       motion       libmysqlclient20 &&     sed -i -e "s/^\(motion:[^:]*\):[0-9]*:[0-9]*:\(.*\)/\1:0:0:\2/" /etc/passwd &&     sed -i -e "s/^\(motion:[^:]*\):[0-9]*:\(.*\)/\1:0:\2/" /etc/group &&     pip install /tmp/motioneye &&     rm -rf /tmp/motioneye &&     apt-get purge --yes python-setuptools python-wheel &&     apt-get autoremove --yes &&     apt-get --yes clean && rm -rf /var/lib/apt/lists/* && rm -f /var/cache/apt/*.bin:
#7 0.590 Get:1 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
#7 0.663 Get:2 http://snapshot.debian.org/archive/debian/20210925 sid InRelease [165 kB]
#7 0.809 Get:3 http://deb.debian.org/debian buster InRelease [122 kB]
#7 0.826 Get:4 http://security.debian.org/debian-security buster/updates/main amd64 Packages [305 kB]
#7 0.970 Get:5 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]
#7 1.186 Get:6 http://deb.debian.org/debian buster/main amd64 Packages [7907 kB]
#7 1.808 Get:7 http://snapshot.debian.org/archive/debian/20210925 sid/main amd64 Packages [8780 kB]
#7 3.804 Get:8 http://deb.debian.org/debian buster-updates/main amd64 Packages [15.2 kB]
#7 7.191 Get:9 http://snapshot.debian.org/archive/debian/20210925 sid/non-free amd64 Packages [112 kB]
#7 7.496 Get:10 http://snapshot.debian.org/archive/debian/20210925 sid/contrib amd64 Packages [61.6 kB]
#7 9.133 Fetched 17.6 MB in 9s (2048 kB/s)
#7 9.133 Reading package lists...
#7 12.81 Reading package lists...
#7 15.86 E: The value 'stable' is invalid for APT::Default-Release as such a release is not available in the sources
------
executor failed running [/bin/sh -c echo "deb http://snapshot.debian.org/archive/debian/$(date +%Y%m%d) sid main contrib non-free" >>/etc/apt/sources.list &&     apt-get update &&     DEBIAN_FRONTEND="noninteractive" apt-get -t stable --yes --option Dpkg::Options::="--force-confnew" --no-install-recommends install       curl       ffmpeg       libmicrohttpd12       libpq5       lsb-release       mosquitto-clients       python-jinja2       python-pil       python-pip       python-pip-whl       python-pycurl       python-setuptools       python-tornado       python-tz       python-wheel       v4l-utils &&     DEBIAN_FRONTEND="noninteractive" apt-get -t sid --yes --option Dpkg::Options::="--force-confnew" --no-install-recommends install       motion       libmysqlclient20 &&     sed -i -e "s/^\(motion:[^:]*\):[0-9]*:[0-9]*:\(.*\)/\1:${RUN_UID}:${RUN_GID}:\2/" /etc/passwd &&     sed -i -e "s/^\(motion:[^:]*\):[0-9]*:\(.*\)/\1:${RUN_GID}:\2/" /etc/group &&     pip install /tmp/motioneye &&     rm -rf /tmp/motioneye &&     apt-get purge --yes python-setuptools python-wheel &&     apt-get autoremove --yes &&     apt-get --yes clean && rm -rf /var/lib/apt/lists/* && rm -f /var/cache/apt/*.bin]: exit code: 100
```

It may be related to the release of Debian bullseye and since this time stable linked to bullseye instead of buster

Adding @kleini who was marked as maintainer of Dockerfile

starbasessd, 2021-09-25:
Will check and confirm.

yurnov, 2021-09-25:
Tried to fix in on my end by changing lines 21-23 of Dockerfile to following:

```
RUN echo "deb http://snapshot.debian.org/archive/debian/$(date +%Y%m%d) sid buster main contrib non-free" >>/etc/apt/sources.list && \
    apt-get update && \
    DEBIAN_FRONTEND="noninteractive" apt-get -t buster --yes --option Dpkg::Options::="--force-confnew" --no-install-recommends install \
``` 

The issue mentioned above is gone, but still build failed due to missing candidate for libmysqlclient20. To fix that I'd tried to change libmysqlclient20 to libmysqlclient21 or libmysqlclient-dev, in both cases I got the following:
```
#8 154.4 /usr/bin/perl: error while loading shared libraries: libcrypt.so.1: cannot open shared object file: No such file or directory
#8 154.4 dpkg: error processing package libc6:amd64 (--configure):
#8 154.4  installed libc6:amd64 package post-installation script subprocess returned error exit status 127
#8 154.4 Errors were encountered while processing:
#8 154.4  libc6:amd64
#8 154.5 E: Sub-process /usr/bin/dpkg returned an error code (1)
```

yurnov, 2021-09-29:
Seems it related to following bug report https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=974552

starbasessd, 2021-09-29:
Interesting thread. 
It may be (at least for now) necessary to freeze the docker container to buster, (and python 2.7) for now. I haven't forgotten this issue, just have a lot on the plate. I will be trying to modify the docker build instructions to take into account some of the errors and upgrades. Since motioneye itself hasn't had any updated work released recently, it's not a real high priority for me, unfortunately.

yurnov, 2021-10-01:
I have successful build docker image with this version of Dockerfile https://github.com/yurnov/motioneye/blob/2e2887db4d8299acd1559bbffe5c5d44f9bb2269/extra/Dockerfile.

I will test on my server and then create a PR

yurnov, 2021-10-03:
Fis was tested and PR created, the issue can be closed after merge

yurnov, 2021-10-06:
The issue is fixed and the ticket can be closed

closed 2021-10-06 by yurnov