motioneye-project/motioneye #2194: SMTP auth: special characters in password field

goooroooX, 2021-09-25:
# Problem
> Email notification do not work when special characters are saved at password field (angle brackets in my case)

## Expected behavior

> Special characters are processing properly

## Actual behavior

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 837, in test
    subject=subject, message=message, files=[])
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 58, in send_mail
    conn.login(account, password)
  File "/usr/lib/python2.7/smtplib.py", line 608, in login
    (code, resp) = self.docmd(encode_cram_md5(resp, user, password))
  File "/usr/lib/python2.7/smtplib.py", line 572, in encode_cram_md5
    response = user + " " + hmac.HMAC(password, challenge).hexdigest()
  File "/usr/lib/python2.7/hmac.py", line 75, in __init__
    self.outer.update(key.translate(trans_5C))
TypeError: character mapping must return integer, None or unicode
   ERROR: notification email test failed: character mapping must return integer, None or unicode
```

starbasessd, 2021-09-25:
Does your email server support hashed, tokenized, delimited or encrypted passwords?
"\<" and "\>" are disallowed characters in many systems and applications.

goooroooX, 2021-09-25:
This is not my server, so I have no idea what kind of passwords it may support.
It is definitely support plain-text auth with special characters. And I cannot change the password.

starbasessd, 2021-09-25:
You could try to edit the camera-X.conf file (X being the camera number)
sudo nano /etc/motioneye/camera-X.conf
Towards the end of the file should be a line beginning: on_event_start
Look for your password in that line (it is a VERY long line)
make sure your password shows as '\<P@ssw0rd\>' 
(with the single quotes and your "\<" & "\>"as appropriate)
Save it, and trigger another email via motion.

goooroooX, 2021-09-25:
```
root@a0d7b954-motioneye:/$ ls -al /etc/motioneye
total 20
drwxr-xr-x    2 root     root          4096 Aug 11 14:23 .
drwxr-xr-x    1 root     root          4096 Sep 26 02:00 ..
-rw-r--r--    1 root     root           249 Aug 11 14:23 motion.conf
-rw-r--r--    1 root     root          2858 Aug 11 14:23 motioneye.conf
```
Can't find password or camera settings in those files.

goooroooX, 2021-09-25:
Removed all special characters from password and got the same error when trying to send Test Email.
```
on_event_start /usr/lib/python2.7/site-packages/motioneye/scripts/relayevent.sh 
"/data/motioneye/motioneye.conf" start %t; 
/usr/bin/python /usr/lib/python2.7/site-packages/motioneye/meyectl.pyc sendmail -c /data/motioneye/motioneye.conf 
'smtp.xxxx.xxx' '25' '***accountname_here***' '***password_here***' 'False' 'hass@from.mail.com' 
'personal@gmail.com' 'motion_start' '%t' '%Y-%m-%dT%H:%M:%S' '0'
```
Weird, but looks like this problem is not related to special characters.

starbasessd, 2021-09-26:
I thought you said "It is definitely support plain-text auth with special characters. And I cannot change the password."
It would have been helpful to mention how you were set up from the beginning.
Since it involves HA and their docker wrapper I must ask you to take that out of the equation.
Please install the motionEye version of docker:
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker 
and see if it occurs there.

JamBalaya56562, 2026-11-08:
Closing, as the reporter found the same error with a password without special characters, so they were not the cause, and this was with the Home Assistant add-on of 2021 (motionEye 0.42.1). The email code has changed a lot since; please open a new issue with the error from the log if a test email still fails with a current release.

closed 2026-11-08 by JamBalaya56562