motioneye-project/motioneye #2195: Movement Detection Query

jsavage, 2021-09-28:
I am using motioneye to monitor my roofspace to monitor a known source of water ingress.
I have the set minimal the thresholds as low as I can but I am not convinced that I am catching every movement that is detectable.  Not least because I can see water dripping outside the red bounding box in the debug in an area that is not masked off.

Any one have any suggestions to make this more sensitive in this application? 

Motioneye 0.42.1
Motion version 4.2.2
OS Raspbian10
Motion settings:
[image]
Example video capture and debug

https://user-images.githubusercontent.com/189524/135128399-ffb720f8-de93-48c6-8340-27d302c263ac.mp4

https://user-images.githubusercontent.com/189524/135128421-7839d48f-8a5d-46d3-862e-728c10fa8200.mp4

starbasessd, 2021-09-28:
Noise level is probably too high. 
Probably need despeckle on
Motion Gap, motion before and motion after are probably overlapping, possibly causing race conditions. (Where the previous motion is still being written when the new motion is detected)
Most of my cameras I have at 2, 10, 10, 2 for gap, before, after, and frames. 
Interesting use case.

jsavage, 2021-09-28:
@starbasessd, tnx for commenting.
I don't get many unexplained motion detections. If anything this setup is not quite as sensitive as I am looking for.
Reduce Noise level - OK will give this a go although I am not totally clear what this does. 
I have always assumed that the purpose of despeckle filter is to remove unwanted random noise - eg the occasional rogue pixel so I am not not clear why switching it on might help.
Tnx for your settings.

Use case still evolving...   Noticed today that spider started moving soon after the rain started... Arachnophobes look away now. A further clue that some movement was not being 'detected' - perhaps being treated as noise.  Will report back on this to close the loop.
 

https://user-images.githubusercontent.com/189524/135163336-2ae43861-fdab-465f-962d-55ce5e82d0ca.mp4

https://user-images.githubusercontent.com/189524/135163341-f0d3a30a-d926-494c-8aa9-c82c7d637e41.mp4

starbasessd, 2021-09-28:
Speckle, like noise, is just a type random pixels. I don't have the really high contrast level that you seem to. For me and my cameras, it (the random pixels) happen more in twilight and dark than the bright light. By reducing the noise detection level (or turning Auto Noise On) you are telling the system the amount of 'noise' that is always there, and might not be 'real', reducing the sensitivity.
https://motion-project.github.io/motion_config.html#noise_level
https://motion-project.github.io/motion_config.html#despeckle_filter
A very handy document in general:
https://motion-project.github.io/motion_config.html

JamBalaya56562, 2026-11-08:
Closing, as this was answered: the sensitivity comes from Motion's settings under Motion Detection, mainly Frame Change Threshold, Noise Level (or Automatic Noise Detection) and Despeckle Filter, as starbasessd explained, with Motion's documentation for the details of each. Small drops against a bright background may simply stay under the noise Motion is told to ignore.

closed 2026-11-08 by JamBalaya56562