motioneye-project/motioneye #2196: No auto switch to night mode in a Raspberry Pi OS install

straytachyon, 2021-09-30:
Hi, I have a pi 3 with a fisheye camera (https://www.aliexpress.com/item/32982250982.html 5MP OV5647 Webcam) that comes with two IR lamp.  I used to run motionEyeOS on it and the camera will switch to night mode if the light gets too low, and switch to normal mode when there is enough light.
I have decided to install the latest version of  Raspberry Pi OS, installed motion through apt-get and installed motionEye by following the installation instructions.  After playing with the configurations, the system works pretty much the same as before.  However, there is a problem with the night mode switch.  The camera will not switch to the night mode.  I have to manually change one of the settings through the motionEye web interface (to restart motion?), then the camera will be in night mode

Can you please tell me what is wrong with my system?
Thanks a lot

**motioneye.conf**
conf_path /etc/motioneye
run_path /var/run
log_path /var/log
media_path /media/motioneye
log_level info
listen 0.0.0.0
port 80
motion_control_localhost true
motion_control_port 8080
motion_check_interval 10
motion_restart_on_errors true
mount_check_interval 300
cleanup_interval 3600
remote_request_timeout 10
mjpg_client_timeout 10
mjpg_client_idle_timeout 10
smb_shares false
smb_mount_root /media
enable_reboot false
smtp_timeout 60
list_media_timeout 120
list_media_timeout_email 10
zip_timeout 500
timelapse_timeout 500
add_remove_cameras true
http_basic_auth false

**motion.conf**
# @enabled on
# @normal_password 
webcontrol_html_output on
webcontrol_port 8080
setup_mode off
webcontrol_parms 2
webcontrol_localhost on
camera camera-1.conf

**camera-1.conf**
# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 
# @upload_username 
# @motion_detection on
# @upload_port 
# @upload_location /FrontDoor
# @preserve_movies 30
# @network_username 
# @upload_movie on
# @id 1
# @webcam_server_resize off
# @upload_password 
# @manual_record off
# @upload_method post
# @upload_picture on
# @working_schedule_type outside
# @network_password 
# @upload_service gdrive
# @preserve_pictures 365
# @storage_device custom-path
# @manual_snapshots on
# @network_share_name 
# @upload_enabled on
# @network_smb_ver 1.0
# @working_schedule 
# @clean_cloud_enabled off

ffmpeg_output_movies on
threshold_maximum 0
stream_quality 50
threshold 251942
quality 85
noise_level 31
ffmpeg_output_debug_movies off
pre_capture 1
noise_tune on
smart_mask_speed 0
stream_maxrate 5
output_pictures best
stream_localhost off
ffmpeg_variable_bitrate 75
ffmpeg_video_codec mp4:h264_omx
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
movie_passthrough off
auto_brightness off
stream_port 8081
rotate 0
stream_auth_method 0
threshold_tune off
framerate 2
emulate_motion off
snapshot_filename %Y-%m-%d/%H-%M-%S
despeckle_filter 
snapshot_interval 0
minimum_motion_frames 20
stream_motion off
target_dir /media/motioneye/FrontDoor
text_double on
vid_control_params video_bitrate_mode=0,sharpness=0,saturation=0,auto_exposure_bias=0,power_line_frequency=1,iso_sensitivity=1,h264_level=11,compression_quality=85,h264_i_frame_period=60,blue_balance=1000,scene_mode=0,contrast=0,color_effects_cbcr=32896,auto_exposure=1,exposure_time_absolute=1000,exposure_metering_mode=0,iso_sensitivity_auto=1,rotate=0,brightness=25,video_bitrate=10000000,white_balance_auto_preset=1,color_effects=0,h264_profile=4,red_balance=1000
post_capture 1
lightswitch 0
stream_authentication user:
output_debug_pictures off
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left FrontDoor
picture_filename %Y-%m-%d/%H-%M-%S
locate_motion_style redbox
locate_motion_mode off
videodevice /dev/video0
max_movie_time 0
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
camera_name FrontDoor
event_gap 30
height 1944
mask_file 
width 2592

yurnov, 2021-09-30:
I looked at a description of this camera and seems like this mode switch is a feature of the camera itself/camera driver in Linux, so, looks like it is not a question to motioneye, rather to the camera vendor how this feature is supported on the latest versions of Raspberry Pi OS.

starbasessd, 2021-09-30:
As you have discovered, any auto switching between daytime (no IR) to night time (IR) modes are usually handled by the camera itself. Were there any other changes / additions made (like HA or external IR lights or accessories)? Maybe additional software? If so, I would check the command:
vcgencmd get_throttled
to see if it is overheating or going under-voltage.

straytachyon, 2021-09-30:
Thanks guys. There was no hardware change. The system was able to switch to the night mode automatically when it was running MotionEyeOS.

vcgencmd get_throttled
throttled=0x70000

Arm frequency capped has occurred
Throttling has occurred
Soft temperature limit has occurred

I suspect the throttling was caused by video encoding though.

starbasessd, 2021-09-30:
It is overheating. This causes all sorts of problems. What other software are you adding that is running on RaspberryPiOS that wasn't available in motionEyeOS? What changed? If you switch back to motionEyeOS, does the camera switch normally? What does the
vcgencmd get_throttled 
report then?

straytachyon, 2021-09-30:
I used the Raspberry Pi OS Lite image and installed only motion and motioneye.
cat /etc/debian_version
10.10

motion version is 4.1.1

Under MotionEyeOS, the video was encoded using the h264/OMX codec while the current setup is using the h264 codec (i assume that because i see this in the motion.log: ffmpeg_set_codec: Preferred codec h264_omx has been blacklisted)

straytachyon, 2021-09-30:
Oh, motion does crash a lot in the new setup.  I assume it crashed because the camera feed stop and if i changed one setting in the MotionEye web interface, the camera feed came back

starbasessd, 2021-09-30:
Sounds like it was maybe 'borderline' in motionEyeOS, and fails in RPiOS (thermally speaking), or coincidentally failed when you switched. 
Your changing a setting and the system forcing a restart to get it back up again is an indication of the thermal throttling. 
Try finding a way to cool the case, put in heat sinks, etc. 
You could try another Pi3B, 3B+ or 4B. 
I don't know which instructions you used to install motion, but the latest instructions install motion 4.2.2, and 4.3.2 is available and tested working. Might be more 'efficient'...

straytachyon, 2021-09-30:
the Debian repo only has motion 4.1.1 i guess.

starbasessd, 2021-09-30:
The Raspberry Pi OS install instructions do not use the Raspbian repo or the Debian repo, it uses the motion github:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb 
dpkg -i pi_buster_motion_4.2.2-1_armhf.deb
To get 4.3.2 just change the 4.2.2-1 to 4.3.2-1

straytachyon, 2021-09-30:
Installed motion 4.3.2 as instructed.  Let's see if the auto switch to IR mode will work tonight

Thank you very much for your help

straytachyon, 2021-09-30:
Screenshot at dusk
https://drive.google.com/file/d/1ajMWEtZoGAco2Gts9H3PJKCoSVTaf9pc/view?usp=drivesdk

IR mode had not turn on yet. I know that because the camera will switch to IR mode with a “click” if I block the light sensor
Auto brightness is off. Turning it on will make the image fluctuate between higher and lower whiteness with zero image improvement

straytachyon, 2021-10-01:
Things are worst with motion 2.3.2. Restarting motion by changing settings on the web interface or restarting the motioneye service will not switch to IR mode.

starbasessd, 2021-10-01:
Did you try to re-load motionEyeOS? Which version were you using before?

straytachyon, 2021-10-01:
Not yet.  I used the latest version of motioneyeos.  Don't remember the exact version

I removed the camera and add it back as before (V4L2, mml something type) and the camera is in IR mode now

starbasessd, 2021-10-01:
CSI ribbon cable cameras are always MMAL, not V4L2.

straytachyon, 2021-10-01:
The camera didn't switch from IR mode back to normal mode this morning.  The image is almost all white due to excessive high ISO. Iso Sensitivity Auto was set to true

starbasessd, 2021-10-01:
Do a clean install of the camera. Do not change ANY settings for the camera (I think you are editing the camera-1.conf file directly) Find out if it is changing mode (night/day) correctly. If it is, then change one item at a time, until you find out which one is not supported on your camera, which one breaks the switch from night to day.

straytachyon, 2021-10-01:
Nope, I never edited the camera config directly.  I made all changes through the web interface

starbasessd, 2021-10-01:
Where is: Iso Sensitivity Auto was set to true.

starbasessd, 2021-10-01:
MotionEyeOS and Fast Network Camera set?
What is the Motion, motionEye, and OS versions under Settings, General?

straytachyon, 2021-10-01:
https://drive.google.com/file/d/1797Nwq_M3M9DxoFCiu53z_IWI3MAO2_J/view?usp=sharing

straytachyon, 2021-10-01:
Not fast network camera, the camera is what I bought from AliExpress and it is connected directly to the Pi with the FFC cable.

starbasessd, 2021-10-01:
I don't know. None of my V4L2 cameras give me those options in MotionEye. I only get those options in motionEyeOS with Fast Network Camera on MMAL type cameras.
What do you get when you delete the camera, re-install it as MMAL type instead of V4L2? 
What do you get for:
vcgencmd get_camera

straytachyon, 2021-10-01:
supported=1 detected=1

straytachyon, 2021-10-01:
I added the camera with camera type "Local V4L2 Camera" and camera "mml service something"

starbasessd, 2021-10-01:
Again, What do you get when you delete the camera, re-install it as MMAL type instead of V4L2?
Delete existing setup
Install new camera
Select MMAL instead of V4L2 in dropdown box

straytachyon, 2021-10-01:
https://drive.google.com/file/d/1c7SP1YOoV-y6WA_pQZkODGtebV-Enj-n/view?usp=sharing

starbasessd, 2021-10-03:
So what was the result of adding as MMAL? Did it auto-switch?

straytachyon, 2021-10-03:
I was thinking of testing for a few more days because motion kept on crashing (the image stuck), but I suppose it did switch mode with the MMAL type selection

closed 2021-10-03 by straytachyon

starbasessd, 2021-10-03:
I would bet, if you can ssh into it, that 
vcgencmd get_throttled
shows a non 0x0 result, and the logs show high temperatures.

straytachyon, 2021-10-03:
throttled=0x20000
Binary: 100000000000000000
under-voltage has occurred since last reboot

Weird, I’m using the google nexus player power supply. Should be enough

starbasessd, 2021-10-03:
Barrel connector? That's 12V. The Google Nexus Player power supply isn't designed for 5.1v, 3amp power.
Using vcgencmd to measure the voltage doesn't really help, unlike doing the temps. The get_throttling command lets you know what the system has done to self-protect.
Try a different power supply, or power cord.

straytachyon, 2021-10-03:
Forgot to thank you for your help.  Thanks a lot

straytachyon, 2021-10-03:
I’ll try to find another power supply then

starbasessd, 2021-10-03:
BTW, phone/tablet chargers, even if rated correctly, seldom can keep the voltage up when pulling anywhere near their full rated current (hence the voltage drop and throttling...)
You're welcome, that's what I'm here for, my payback to CCrisan and the community.