motioneye-project/motioneye #2198: No SMB option with RTSP camera

valemaio2, 2021-10-02:
I have MotionEye running as a Docker container on a Lubuntu 18.10 server. MotionEye version is 0.42, motion version is 4.2.2.

Using other MotionEyeOS cameras running on Raspberry Pis I can set up the main MotionEye to save movies on a Samba server (running on the same metal hardware but on a different container); however, I don't have the option for Samba when using an RTSP camera (see picture attached). 

Is this by design or am I missing something? Happy to provide any logs if I need to.
[image]

starbasessd, 2021-10-02:
Docker, like motionEye on another OS, depends on the OS to handle SMB. You would mount the SMB Share, and use a -v (volume) link in the docker run command.
I use:
in /etc/fstab in the base OS:
//ip_address/share_name /media/share_name cifs vers=3.0 credentials=/home/username/.credentials
in the .credentials file:
username=[username]
password=[password]
without the brackets
add 
-v /media/share:/media/share
then use Custom Path and modify Root to
/media/share
Modify as needed for your use case.
The only system with SMB services built in is motionEyeOS... The other services (FTP, SFTP, GDrive, Dropbox) are available in Uploads, but, IIRC it hasn't been updated for the latest DropBox and GDrive issues...

valemaio2, 2021-10-02:
> The only system with SMB services built in is motionEyeOS

Yeah, that actually makes a lot of sense. 

My bare metal /etc/fstab already had the mount I wanted to use, so all I had to do was adding the starting option `-v /media/share:/media/share` and BAM, it's back recording. Thank you very much.

closed 2021-10-02 by valemaio2