motioneye-project/motioneye #2201: "hidden" camera

enry86cami, 2021-10-07:
Hi all,
in the motioneye webpage I see 3 cameras, and it is ok.
[image]

But from http://myip:7999 I see 4 cameras:
[image]

Is it normal? How could I remove the first camera?

enry86cami, 2021-10-07:
> What happens when you click on 'Camera 1 and then click List?

Here you are:
`Camera 1
setup_mode = off
camera_name = (not defined)
videodevice = /dev/video0
v4l2_palette = 17
input = -1
norm = 0
frequency = 0
flip_axis = none
rotate = 0
width = 352
height = 288
framerate = 2
minimum_frame_time = 0
netcam_url = http://192.168.0.19/tmpfs/auto.jpg
netcam_highres = (not defined)
netcam_userpass = admin:Audi.a.3!!
netcam_keepalive = off
netcam_proxy = (not defined)
netcam_tolerant_check = on
mmalcam_name = (not defined)
mmalcam_control_params = (not defined)
auto_brightness = off
brightness = 0
contrast = 0
saturation = 0
hue = 0
power_line_frequency = -1
roundrobin_frames = 1
roundrobin_skip = 1
switchfilter = off
threshold = 7680
threshold_tune = off
noise_level = 31
noise_tune = on
despeckle_filter = (not defined)
area_detect = (not defined)
mask_file = /etc/motioneye/mask_2.pgm
mask_privacy = (not defined)
smart_mask_speed = 0
lightswitch = 0
minimum_motion_frames = 2
pre_capture = 4
post_capture = 4
event_gap = 2
max_movie_time = 0
emulate_motion = off
output_pictures = off
output_debug_pictures = off
quality = 85
camera_id = 1
picture_type = jpeg
ffmpeg_output_movies = off
ffmpeg_output_debug_movies = off
ffmpeg_bps = 400000
ffmpeg_variable_bitrate = 75
ffmpeg_video_codec = mp4:h264_omx
ffmpeg_passthrough = off
ffmpeg_duplicate_frames = off
use_extpipe = off
snapshot_interval = 0
locate_motion_mode = off
locate_motion_style = redbox
text_right = %Y-%m-%d\n%T
text_left = (not defined)
text_changes = off
text_event = %Y%m%d%H%M%S
text_double = off
exif_text = (not defined)
target_dir = /media/pi/CAE5-3804/ext
snapshot_filename = %v-%Y%m%d%H%M%S-snapshot
picture_filename = %v-%Y%m%d%H%M%S-%q
movie_filename = %Y-%m-%d/%H-%M-%S
timelapse_interval = 0
timelapse_mode = daily
timelapse_codec = mpeg4
timelapse_fps = 30
timelapse_filename = %Y%m%d-timelapse
ipv6_enabled = off
stream_port = 8082
substream_port = 0
stream_quality = 85
stream_motion = off
stream_maxrate = 5
stream_localhost = on
stream_limit = 0
stream_preview_scale = 25
stream_preview_newline = off
track_type = 0
track_auto = off
track_port = (not defined)
track_motorx = 0
track_motorx_reverse = off
track_motory = 0
track_motory_reverse = off
track_maxx = 0
track_minx = 0
track_maxy = 0
track_miny = 0
track_homex = 128
track_homey = 128
track_iomojo_id = 0
track_step_angle_x = 10
track_step_angle_y = 10
track_move_wait = 10
track_speed = 255
track_stepsize = 40
quiet = on
sql_log_picture = on
sql_log_snapshot = on
sql_log_movie = off
sql_log_timelapse = off
sql_query_start = sql_query_start insert into security_events(camera, event_time_stamp) values('%t', '%Y-%m-%d %T')
sql_query = sql_query insert into security_file(camera, filename, frame, file_type, time_stamp) values('%t', '%f', '%q', '%n', '%Y-%m-%d %T')
database_type = (not defined)
database_dbname = (not defined)
database_host = localhost
database_port = 0
database_busy_timeout = 0
video_pipe = (not defined)
motion_video_pipe = (not defined)`

starbasessd, 2021-10-07:
What happens when you click on 'Camera 1 and then click List?

starbasessd, 2021-10-07:
Docker or direct install? Linux? 
The Camera 1 is showing /dev/video0 which usually is only on a Linux box with a camera direct connected.

enry86cami, 2021-10-07:
` motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion; generated; vendor preset: enabled)
   Active: active (exited) since Mon 2021-09-27 11:30:32 CEST; 1 weeks 3 days ag
     Docs: man:systemd-sysv-generator(8)
  Process: 363 ExecStart=/etc/init.d/motion start (code=exited, status=0/SUCCESS
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/motion.service

set 27 11:30:30 raspberrypi systemd[1]: Starting LSB: Start Motion detection...
set 27 11:30:33 raspberrypi motion[363]: Not starting motion daemon, disabled vi
set 27 11:30:32 raspberrypi systemd[1]: Started LSB: Start Motion detection.

`

starbasessd, 2021-10-07:
One of your config files is trashed. You can try copying the default files back:
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
and re-add your cameras.

starbasessd, 2021-10-07:
Do you have a folder:
/etc/motion
??? What is in it?

starbasessd, 2021-10-07:
Copy the sample file back again, reboot, do not add the cameras and check port 7999.
Port 7999 should not be active.

enry86cami, 2021-10-07:
how can I perform a clean uninstallation?

starbasessd, 2021-10-07:
You really can't. 
You could try going to the install instructions, and running 'sudo apt purge' followed by all the programs installed, but you said you did it a while back (the install) and it may have been different than the current version. Also, doing it that way has a tendency to remove stuff other programs may need. 
Maybe someone else has ideas, I'll leave it here open.

enry86cami, 2021-10-08:
thanks for your support.... I can live with this problem....

JamBalaya56562, 2026-11-10:
Closing with the answer: the first entry on Motion's web control page (port 7999) is Motion's main `motion.conf`, camera 0, which stands for all cameras, not a camera of its own, so with three cameras it lists four. Nothing needs removing.

closed 2026-11-10 by JamBalaya56562