motioneye-project/motioneye #2202: Network Failed Panic Rebooting

cantercrow, 2021-10-09:
Good day,

I have a Raspberry PI 4B 8G with a DSLR KIT Power Over Ethernet PoE HAT IEEE802.3af DC 5V 2.5A with 1.5KV Isolation for Raspberry Pi 4B 3B+ 3B Plus

If I try and boot MotionEye using POE I get the error "Network Failed Panic Rebooting", I have tried with DHCP and Static.

If I boot without POE I don't get the error.

Raspbian boots ok from POE so not certain what the issue is.

cantercrow, 2021-10-09:
[boot.log](https://github.com/ccrisan/motioneye/files/7316117/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneye/files/7316118/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneye/files/7316119/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/7316120/motioneye.log)

starbasessd, 2021-10-09:
Which version of motionEyeOS? You need dev20201026 for Pi4b-8GB.

cantercrow, 2021-10-09:
motioneyeos-raspberrypi3-20200606 & motioneyeos-raspberrypi4-dev20201026

starbasessd, 2021-10-09:
It's saying No Link. Does it patch the ethernet to the on-board? If not, motionEyeOS is not seeing the ethernet port.

cantercrow, 2021-10-09:
https://www.amazon.co.uk/gp/product/B07JPXR9ZN/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1

starbasessd, 2021-10-09:
If the Pi4-8GB is V1.1 (check the label on the board (between the GPIO pins and the CPU) then it is a known issue that was resolved in the RPiOS sometime after Dec 2020, but we haven't seen a new build since Oct 2020...

starbasessd, 2021-10-09:
Which is why it works with the latest RPiOS but not with the latest dev version of motionEyeOS. You might be able to do a 'build your own' using the instructions on the motionEyeOS Wiki:
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
My discoveries:
https://github.com/raspberrypi/linux/issues/3108
https://github.com/raspberrypi/linux/issues/3034
https://github.com/raspberrypi/linux/issues/3195
Although they state 'after reboot' they also state on 'power up', too...

JamBalaya56562, 2026-11-08:
Closing, as this is about motionEyeOS on a Pi 4 with a PoE HAT, where the kernel of the image does not see the network link, while Raspberry Pi OS does. That is part of motionEyeOS, which is maintained in its own repository (motioneye-project/motioneyeos).

closed 2026-11-08 by JamBalaya56562