motioneye-project/motioneye #2204: Email Notification ( Motion notification ) Timming out

umairulh2001, 2021-10-11:
I had the motion notification through email working suddenly it started giving timeout errors in the log. Below is the error details from log file. I have also tried different mail servers but i am getting same issue

`Oct 11 12:14:34 raspberrypi meyectl[251]: Traceback (most recent call last):
Oct 11 12:14:34 raspberrypi meyectl[251]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 831, in test
Oct 11 12:14:34 raspberrypi meyectl[251]:     subject=subject, message=message, files=[])
Oct 11 12:14:34 raspberrypi meyectl[251]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/sendmail.py", line 53, in send_mail
Oct 11 12:14:34 raspberrypi meyectl[251]:     conn = smtplib.SMTP(server, port, timeout=settings.SMTP_TIMEOUT)
Oct 11 12:14:34 raspberrypi meyectl[251]:   File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
Oct 11 12:14:34 raspberrypi meyectl[251]:     (code, msg) = self.connect(host, port)
Oct 11 12:14:34 raspberrypi meyectl[251]:   File "/usr/lib/python2.7/smtplib.py", line 316, in connect
Oct 11 12:14:34 raspberrypi meyectl[251]:     self.sock = self._get_socket(host, port, self.timeout)
Oct 11 12:14:34 raspberrypi meyectl[251]:   File "/usr/lib/python2.7/smtplib.py", line 291, in _get_socket
Oct 11 12:14:34 raspberrypi meyectl[251]:     return socket.create_connection((host, port), timeout)
Oct 11 12:14:34 raspberrypi meyectl[251]:   File "/usr/lib/python2.7/socket.py", line 575, in create_connection
Oct 11 12:14:34 raspberrypi meyectl[251]:     raise err
`

starbasessd, 2021-10-11:
I just tested my Ubuntu 20.04 (latest updates), i5/8GB/1TB, stand alone (no other software) to GMail, and it is working normally.
When you go to Motion Notifications, click on Test Email, what result do you get? (Short pop-up at the top)
What has changed since it last worked? Any new software? Any updates?

umairulh2001, 2021-10-11:
I am using it with Raspberry , It is very weird previously i tested multiple times and i was getting timeout error. But now after your message when i check for what was the error message it started working correctly.

It might be possible that the smtp server of email side might have some issue at the time. It is working now

starbasessd, 2021-10-11:
Someone may have been doing patches and updates, Sundays are a common day to do so. If they don't have a number of servers (Like Google or Outlook or major Enterprises) then they'd be depending on sending servers to spool and cache for when they get done... BTDT as an admin.