motioneye-project/motioneye #2206: `An error occurred. Refreshing is recommended` when using custom 404 or 403 pages

kdheepak, 2021-10-17:
I'm using `traefik` as a reverse proxy, and when using custom error pages, it overrides the 404 and 403 pages for any service it runs. This means when motioneye requests for `/login` or `/config/list` and expects 403 with a json but it gets a custom html error page. This causes the following error:

[image]
 
I understand if you think this is not a motioneye issue, since everything works fine when I don't use custom error pages. I just wondering if you can make a change to not fail for invalid 403 errors?

kdheepak, 2021-10-17:
Here’s the link to the related issue on traefik: https://github.com/traefik/traefik/issues/4218

motioneye is the only service ( among 5 others that I am using ) that fails when using a custom error page.

kdheepak, 2021-10-17:
I think “take it to traefik, not our problem” is perfectly reasonable btw. 

The reason I want custom error pages is that I currently have multiple users that I expect will sign in to use the services on my raspberrypi, but only some services will be available to some users. It would be nice to display a custom error page telling them they don’t have access to a page instead of traefik’s default error page which is basically just a single word in an unformatted html page, e.g. `forbidden` for 403. 

This is probably just a traefik issue, but there’s no easy way for me to solve it there and I was hoping that make a change here could help me or even a patch that I can apply locally could help me. 

I would like motioneye to not reply on the json returned by error code pages. If I’m able to make that patch I expect everything will work for me fine. This is probably very specific to me, so I understand if you are not interested in this at all and want to just ignore this help request :) I don’t expect it but I would super appreciate it. I tried to look through the code base but got overwhelmed fairly quickly.

starbasessd, 2021-10-17:
Why not use a simple re-direct for motioneye.domain.com/login/ to motioneye.domain.com similar to www.motioneye.domain.com usually redirects to motioneye.domain.com \(or reverse\)?

kdheepak, 2021-10-17:
The problem is that there is javascript code inside this project that makes a request to `/login/?...` and expects `{prompt: true, error: "unauthorized"}` when the `...` is an anonymous or unauthorized user. Similarly there are calls to `/config/lists`. Redirecting this to `motioneye.domain.com` does not work to solve this issue. Does that make sense?

starbasessd, 2021-10-17:
Yep, makes sense. Need to figure out how to change the send 'expect' javascript on the traefik  side. It's an interesting issue. I'll re-open it for you, and we'll leave it here to see if anyone has any suggestions, and I might dig a bit deeper into the python code to see if we can get something other than the json page...
BTW, it happens in all versions, motionEye in docker, motionEye on another OS, and motionEyeOS. It happens whether it's /login, or /purple. It's the response the python code gives to unexpected data in the URL, I believe...

kdheepak, 2021-10-18:
Making that changes seems to resolve the original issue for me, and everything else seems to work as expected!

MichaIng, 2022-03-18:
Solved in current `dev` branch. Marking this as closed.

PR #2211 (2022-03-18): Change `onResponse` argument to avoid dependency on error pages
Fixes #2206
closed 2022-03-18 by MichaIng