motioneye-project/motioneye #2207: Multiple Motion Threads

dbeckwith01, 2021-10-17:
I am evaluating MotionEye to use to monitor multiple IP video cameras around my home.  I hoping to run this on my RPI 4B+.  I have successfully got MotionEye setup with one RTSP camera but I noticed it's using ~40% of the CPU.  Running htop shows a large number of motion process which seem to run across multiple cores.  I was hoping that MotionEye would set motion up to use the RPi's hardware encoding.

It is possible to get MotionEye to use the RPI hardware encoder?

[image]

dbeckwith01, 2021-10-18:
@starbasessd,

Thank you for the quick reply!  

Currently I'm not running MotionEye with the Motion settings.  I'm just testing streaming and recording.  I'm running Motion in Docker at the moment on Balena OS.

I'm using MotionEye 0.42.1 and Motion 4.3.2.

[image]

JamBalaya56562, 2026-11-08:
Closing, as this was answered: Motion has to decode every frame on the CPU to look for motion, which a hardware encoder cannot take over, and the many motion entries in htop are the threads of that one process. The Pi's encoder helps only for writing movies: Movies > Movie format offers H.264/V4L2M2M (.mp4) when the installed ffmpeg supports it, and Movie Passthrough saves the camera's own stream without encoding at all, at the cost of the text overlay.

closed 2026-11-08 by JamBalaya56562