motioneye-project/motioneye #2209: motioneye in docker motion detection menu missing?

boyphongsakorn, 2021-10-18:
i install motioneye in docker this work but many menu is missing (like motion detection,webhook) (i login on admin)
[url=https://ibb.co/HKcqmzk][img]https://i.ibb.co/HKcqmzk/image.png[/img][/url]
how i fix this or motion detection not work on ccrisan/motioneye:master-armhf

starbasessd, 2021-10-20:
@Korte68 How about some more info? 
BaseOS? Which version of docker? What is your docker run command?

Korte68, 2021-10-20:
Home assistant core add-on install.

starbasessd, 2021-10-20:
Just threw RaspberryPiOS Lite on a Pi3B+
Ran 
sudo apt update && sudo apt upgrade -y
sudo reboot
sudo apt install docker.io -y
sudo usermod -aG docker pi
sudo reboot
docker run --name="motioneye" \\
-p 8765:8765 \\
--hostname="motioneye" \\
-v /etc/localtime:/etc/localtime:ro \\
-v /etc/motioneye:/etc/motioneye \\
-v /var/lib/motioneye:/var/lib/motioneye \\
--restart="always" \\
--detach=true \\
ccrisan/motioneye:master-armhf 
Opened web browser on Win10
Connected to http://ip_address:8765
Added Network camera \(d-link DCS-5020L\)
Screenshots showing Motion Detection, and both locations with Web Hooks.
[image]
[image]
[image]
All functioning normally.
I would suggest setting up a separate docker, or direct install, if you don't get any help from HA Community.

closed 2021-10-20 by starbasessd