motioneye-project/motioneye #2212: detection active on restart

enry86cami, 2021-10-19:
Hi,
I installed motioneye on raspberry with raspbian.
I have the camera n°4 configured with movie ON when motion triggered. Motion detection is switched off. I performed the configuration by webpage.

I start/stop recording by commands:
curl "http://ip:7999/4/config/set?emulate_motion=1"
curl "http://ip:7999/4/config/set?emulate_motion=0"
And them work.

When I restart motioneye or when I apply a modify in webpage, the camera 4 detection goes to "active" and the recording starts...

Can I avoid this?

starbasessd, 2021-10-19:
Sounds like the 4th camera isn't quite set up like the others. I would compare /etc/motioneye/Camera-4.conf with the other 3 files.

enry86cami, 2021-10-20:
> I would compare the settings in the WebGUI, specifically in Motion Detection, between the 2 cameras. The IDs don't match the web stream ports (usually they are id 1 & 8081, id 2 and 8082, etc). This indicates something is 'broken' with the editing (either manually or with WebGUI, and / or several additions, deletions, re-additions. You said Camera 4 was the one broken, but you are showing cameras 3 & 2 here...

Yes I done several additions, deletions an re-addition... The camera 4 is ID3

starbasessd, 2021-11-08:
Closing, no indication of user checking files or configs. If unresolved, feel free to open a new issue.

closed 2021-11-08 by starbasessd