motioneye-project/motioneye #2213: adding python modules for MotionEyeOS

davidbroster, 2021-10-21:
Hello, I haven't figured which Linux distro MotionEyeOS uses ..... I was hoping to run some python code but since there's no PIP I don't know how to add any modules .....

I've also tried MotionEyes on top of PiOS (lite) but it doesn't work as well as it does on your OS .....

Any advice please ?

starbasessd, 2021-10-21:
motionEyeOS runs on buildroot/thingOS. Wiki here:
https://github.com/ccrisan/motioneyeos/wiki
it doesn't have the ability to add stuff unless it's stand alone (like scripts). It doesn't have pip, apt, or other similar programs. Most of the system is Read Only.
It only has enough OS to run the motionEye python setup.

davidbroster, 2021-10-31:
> With the exception of Expert Settings, what else is 'missing'? Certain functionality is passed back to the BaseOS in motionEye. (Such as access to shared drives, etc)
> You can use motionEye (say, in an Ubuntu VM) to extract and modify the .pyc files, then re-compile them back to .pyc, then upload them to motionEyeOS. Kinda twisted method, but you have to remember the motionEyeOS is designed to be 'safe' (Read Only filesystem for the most part, overlays for the rest) and is quite sophisticated overall.
> As to the USB camera issue, I have 2 'Generic' Tiga webcams that I just use one on a Windows Box because it doesn't like any RPi board. Same Make, Model, Vendor ID, etc. Just grumpy.

Hi Calin,

I finally have a stable system with both camera’s running.

You asked what is missing other than expert-settings, its not what’s missing more that “ONLY” using motioneyeos do I manage to actually detect and connect to both my camera’s.
In my one camera system I also had a simply python/bottle script that I could use to set the brightness in the bordbox using the in-camera LED and a PWM GPIO.  It would be lovely if I could assign a button to PWM my LED at 0% 25% 50% 100% brightness.  But as far as I’ve read there’s no PWM module in the OS.

My current solution is to run MotionEyeOS on a pi3 and implement the lighting controls on a separate piZero ….. a bit messy but workable.

I’m not expert enough to add modules to your OS or to rebuild with the extra bits in.

I just got a Pi Zero 2W and can boot it using the Pi3 image but there’s no Wifi detected …..

I do love MotionEyeOS though ….. primarily because it loves my endoscope cameras 😊

David

________________________________

David & Margaret Broster
“Greenewood”, The Park, Great Barton, IP31 2SX, Suffolk, United Kingdom

GPS Coordinates:   N52°16'14" E00°45'45" (52.270758, 0.762636)    ***@***.***,0.762636

House:               +44 (0)1284 788 255
Mobile David         +44 (0)7557 359 095     ***@***.******@***.***>
Mobile Margaret    +44 (0)7557 359 417     ***@***.******@***.***>

JamBalaya56562, 2026-11-08:
Closing, as this is about adding software to motionEyeOS, which is a read-only Buildroot system without pip, maintained in its own repository (motioneye-project/motioneyeos). On Raspberry Pi OS, motionEye is installed with pip and other Python modules can be added the same way.

closed 2026-11-08 by JamBalaya56562