motioneye-project/motioneye #2218: [GUI] Can not remove cam

ZoneTwelve, 2021-10-30:
I trying to remove the unavailable camera, but ... yet.
It's happened when I reboot the networking, then the entire Interface is broken.

[image]

starbasessd, 2021-10-30:
How did you remove the camera?

starbasessd, 2021-10-30:
Might be helpful to know which OS, Which set of instructions used to install with, anything like docker, HA, HASS, involved, etc.

ZoneTwelve, 2021-10-31:
Linux CAM00 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
   Static hostname: CAM00
         Icon name: computer
        Machine ID: 36b77e...
           Boot ID: a236337...
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.10.17-v7+
      Architecture: arm

Install method: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

And, because my camera is not working, so I just trying to remove it from Web-GUI

starbasessd, 2021-10-31:
Looks like the motion.conf or motioneye.conf file is corrupted (/etc/motioneye/motion.conf or /etc/motioneye/motioneye.conf) as you are missing a trash can icon to the right of the camera name drop down list. 
You can try to restore the default motioneye.conf file:
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample \\
 /etc/motioneye/motioneye.conf

and reboot.
If that doesn't work, you may have to start over.

ZoneTwelve, 2021-11-02:
> Looks like the motion.conf or motioneye.conf file is corrupted (/etc/motioneye/motion.conf or /etc/motioneye/motioneye.conf) as you are missing a trash can icon to the right of the camera name drop [...]

Thx for the solution you have given, it's working right now.
Really helpful.

closed 2021-11-02 by ZoneTwelve