motioneye-project/motioneye #2221: Not recording to NAS

peterd550, 2021-11-03:
I now have 2 camera in motion eye
Both working
motionEye Version | 0.42.1
Motion Version | 4.3.1
OS Version | motionEyeOS dev20201026

The camera can connect to the samba share and create a folder but save any video even though continues recording is selected

Please see logs
[dmesg.log](https://github.com/ccrisan/motioneye/files/7464995/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneye/files/7464996/messages.log)
[motion.log](https://github.com/ccrisan/motioneye/files/7464997/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/7464998/motioneye.log)
gs

starbasessd, 2021-11-03:
At first glance, it looks like the motionEye config for the smb share is incorrect. 
2021-11-03 11:45:41: [motioneye]    ERROR: access to network share //\\raspiraid\/motioneye1\garden failed: cannot mount network share
2021-11-03 11:47:20: [motioneye]    ERROR: failed to mount smb share "//raspiraid/motioneye1" at "/data/media/motioneye_raspiraid_motioneye1"
2021-11-03 11:47:20: [motioneye]    ERROR: access to network share //raspiraid/motioneye1 failed: cannot mount network share
2021-11-03 11:50:10: [motioneye]    ERROR: failed to mount smb share "//raspiraid/motioneye1" at "/data/media/motioneye_raspiraid_motioneye1_movies"
2021-11-03 11:50:10: [motioneye]    ERROR: access to network share //raspiraid/motioneye1 failed: cannot mount network share
2021-11-03 11:51:51: [motioneye]    ERROR: failed to mount smb share "//raspiraid/motioneye1" at "/data/media/motioneye_raspiraid_motioneye1_movies"
2021-11-03 11:51:51: [motioneye]    ERROR: access to network share //raspiraid/motioneye1 failed: cannot mount network share
2021-11-03 11:53:59: [motioneye]    ERROR: failed to mount smb share "//\\raspiraid/\motioneye1" at "/data/media/motioneye___raspiraid__motioneye1"
2021-11
2021-11-03 12:08:58: [motioneye]    ERROR: failed to unmount smb share "//10.0.0.140/motioneye1\134garden" from "/data/media/motioneye_10_0_0_140_motioneye1_134garden"
2021-11-03 12:10:19: [motioneye]    ERROR: failed to unmount smb share "//10.0.0.140/motioneye1\134garden" from "/data/media/motioneye_10_0_0_140_motioneye1_134garden"
2021-11-03 12:11:21: [motioneye]    ERROR: failed to unmount smb share "//10.0.0.140/motioneye1\134garden" from "/data/media/motioneye_10_0_0_140_motioneye1_134garden"
2021-11-03 12:13:23: [motioneye]    ERROR: failed to unmount smb share "//10.0.0.140/motioneye1\134garden" from "/data/media/motioneye_10_0_0_140_motioneye1_134garden"
2021-11-03 12:13:33: [motioneye]    ERROR: failed to unmount smb share "//10.0.0.140/motioneye1\134garden" from "/data/media/motioneye_10_0_0_140_motioneye1_134garden"
2021-11-03 12:13:54: [motioneye]    ERROR: failed to unmount smb share "//10.0.0.140/motioneye1\134garden" from "/data/media/motioneye_10_0_0_140_motioneye1_134garden"
2021-11-03 12:18:40: [motioneye]    ERROR: failed to unmount smb share "//10.0.0.140/motioneye1\134garden" from "/data/media/motioneye_10_0_0_140_motioneye1_134garden"

This is usually caused by permissions to the share on the SMB host, or Security Version of SMB. I have an older NAS which only supports V1, and I must select the correct version when making connections. 
The little tool tips at the end of each line can be of assistance in getting the correct path. I will post examples in the AM (EDT) when I can access them.

starbasessd, 2021-11-03:
If you are using a Windows SMB host here's a 'suggestion' for 'testing':
https://docs.microsoft.com/en-us/windows-server/storage/file-server/troubleshoot/detect-enable-and-disable-smbv1-v2-v3
The snippets from the log were from the motioneye.log for you to look through.

peterd550, 2021-11-03:
I'm using samba for my nas

It has access to create the folder but will not save any videos in the folder

starbasessd, 2021-11-03:
Samba on what? Debian, Ubuntu, RHEL, which flavor of Linux? Or which brand/model of NAS? Or something like HomeAssistant, OMV, or the like? 
Which version of Security is it using?

peterd550, 2021-11-03:
I'm using raspberian samba with no security
It has read write and quest 

Think it's a bit strange the folders are created but not the files

starbasessd, 2021-11-03:
You might want to check the permissions on the folder tree on the Samba server. Never a 'good idea(tm)' to use guest permissions on any samba server, especially for write access.

peterd550, 2021-11-03:
The samba share is pi:pi but the chmod is 777
It's not protected and is open to everyone

starbasessd, 2021-11-03:
I'll post examples of working setups when I get up in the AM (EDT).

starbasessd, 2021-11-03:
Older Buffalo NAS (SMB V1 Security):
SMB Share is on Primary Share: Share_Drive1
Folder tree used Share_Drive1/pi3 
User & Password valid for user access
- - - - -
Video Device
- - - - -
Camera Name  Camera1
Camera ID  1
Camera Device  http://ip_address:8081
Camera Type  Network Camera
Automatic Brightness  Off
- - - - -
File Storage
- - - - -
Storage Device  Network Share
Network Server  ip_address
SMB Protocol Version  1.0
Share Name  Share_Drive1/pi3
Share Username  username
Share Password  password
Root Directory  /
- - - - - 
Test Share: Successful
- - - - -
Still Images Off
Movies On
- - - - -
Movie File Name  %Y-%m-%d/%H-%M-%S
Movie Passthrough Off
Movie Format  H.264/OMX (.mp4)
Movie Quality  75%
Recording Mode  Continuous Recording
Maximum Movie Length  60 Seconds ***
Preserve Movies  1 day
*** I do not know what length you set to, if left at zero it never fully creates the movie, as it never actually stops to write the file. I recommend no longer than 3600 seconds (1 hour). File sizes at that length can be VERY large. Also, the current timeframe recording cannot be viewed (like scrubbing back the last X minutes to re-run). Only written files can be viewed.
Will show settings for a RPi4-4GB Samba Server set up shortly, have to throw one together.

starbasessd, 2021-11-03:
Created a Pi4-4GB, RaspberryPiOS (fully updated) Samba Server.
Added /share folder with 777 permissions
Added to smb.conf:
[share]
Comment = Pi shared folder
Path = /share
Browseable = yes
Writeable = Yes
only guest = no
create mask = 0777
directory mask = 0777
Public = yes
Guest ok = yes
- - - - -
Changed the following settings in the Pi3:
Storage Device Network Share
Network Server ip_address  *** Pi4 ip address
SMB Protocol Version 1.0
Share Name share/pi3
Share Username guest
Share Password guest
Root Directory /
- - - - -
Clicked "Apply" & "Test"
Was successful.
Files started appearing about a minute later, as expected.

peterd550, 2021-11-04:
Thanks the issue was to set the folder with 777 permissions.
the owner was pi but I still had to sudo
thanks for your help

closed 2021-11-04 by peterd550