motioneye-project/motioneye #2223: Video not showing

Wulmund, 2021-11-04:
I have tried latest Chrome, Edge, even tried IE. I get the ‘thinking’ circle but the videos never load up. They do work in safari on my phone, but I need to change the input mask which I can’t do in Safari. Does anyone know why this might be?

starbasessd, 2021-11-04:
Firefox: Ver 94
Edge: Version 95.0.1020.40 (Official build) (64-bit)
Chrome: Version 95.0.4638.69 (Official Build) (64-bit)
Do your browsers enforce https? Most do. It can be changed.
motionEye is http only
Are you going to http://ip_address:8765? This is the correct URL for motionEye
If you are using motionEyeOS, the URL is http://ip_address
Do you have VLC installed? It does help with playback.
Do you have a SSH program (PuTTY for example on Win10)?
If you ssh into the machine running motionEye, can you change directory to
the folders where video is stored? Anything there?
How much free space on the drive where the videos are stored?

starbasessd, 2021-11-27:
Please respond or close

Wulmund, 2021-11-28:
I ran an update and it resolved this issue

closed 2021-11-28 by Wulmund

starbasessd, 2021-11-28:
May I ask, updated what?