motioneye-project/motioneye #2224: should pytz be listed as a dependency in configuration files?

yuluc123, 2021-11-07:
Hi,
I notice that pytz is used in the files- motioneye/tzctl.py.  But pytz is not listed in any configuration file. So should pytz be added to configuration files as a dependency?
[image]

starbasessd, 2021-11-07:
Interesting.
On my production Hub (Ubuntu 20.04 / install instructions from the WiKi) my system doesn't have pytz. Camera overlays and filenames all show correct time & dates even after the EDT -> EST roll back this morning.  I looked on many distros I have running here (Ubuntu 18.04, 20.04, 21.10, Debian 9, 10, 11, RaspberryPiOS, Custom RPiOS) and only one had pytz (found at ./usr/lib/python3/dist-packages/mypy/typeshed/third_party/2and3/pytz). None of those other machines had any issues this morning, all reporting correct TZs and time after the roll back. 
Was there a symptom you were looking at? It could be to cover an 'esoteric' issue...

starbasessd, 2021-11-27:
Please respond or close.

yuluc123, 2021-11-30:
Sorry to reply so late.
I am just having a start with motioneye. I am confused by the usage of pytz and can not find any explict instructions of pytz. So I open the issue and give suggestion that maybe it should be declared in some configuration files.
But it seems that motioneye use **python-tz** instead of pytz in _extra/dockerfile_? If so, I will close this issue.

starbasessd, 2021-11-30:
I think you answered your own question. If pytz is required, it would be already installed (docker image) or part of the install instructions in the wiki. It may be tested for, but if not found, use a different method.

JamBalaya56562, 2026-11-10:
Fixed by #3464, which replaced the last use of pytz with the standard library's `zoneinfo`, so motionEye no longer needs pytz at all.

closed 2026-11-10 by JamBalaya56562