motioneye-project/motioneye #2225: Unable to make Zoom in / Zoom out work

doucettom, 2021-11-08:
Hi,

I am trying to make an ONVIF cam Action buttons. I've been able to make pan/tilt work using this [how-to](https://www.instructables.com/MOTIONEYE-NVR-PANTILT/) but so far, I've been unable to make the zoom in/zoom out buttons work reliably. Thus, I'd need some help if someone already done this. My camera is ONVIF-compliant, this really looks like a lack of Python coding skills (or, in all honesty, scripting skills in general)

TIA

starbasessd, 2021-11-08:
It might help to post what script(s) you are using for zoom in/out and Action Buttons...
Can you generate the zoom command outside of motionEye?
Does zoom work on your camera in an app?
I have some d-link cameras that were sold as PTZ (pan-tilt-zoom) but were only pan-tilt and zoom was a function of software only.

doucettom, 2021-11-08:
> It might help to post what script(s) you are using for zoom in/out and Action Buttons... Can you generate the zoom command outside of motionEye? Does zoom work on your camera in an app? I have some [...]

`sysadmin@1001-noc-nvr-01:/etc/motioneye$ cat zoom_in_1
#!/usr/bin/env python3
#Delete the none applicable shebang above
#Pan Tilt application for ONVIF devices

from time import sleep
from onvif import ONVIFCamera

ZMAX = 1
ZMIN = -1

def perform_move(ptz, request, timeout):
    # Start continuous move
    ptz.ContinuousMove(request)
    # Wait a certain time
    sleep(timeout)
    # Stop continuous move
    ptz.Stop({'ProfileToken': request.ProfileToken})

def zoom_out(ptz, request, timeout=1):
    request.Velocity.Zoom._x = ZMIN
    request.Velocity.PanTilt._x = 0
    request.Velocity.PanTilt._y = 0
    perform_move(ptz, request, timeout)

def zoom_in(ptz, request, timeout=1):
    request.Velocity.Zoom._x = ZMAX
    request.Velocity.PanTilt._x = 0
    request.Velocity.PanTilt._y = 0
    perform_move(ptz, request, timeout)

def continuous_move():
    mycam = ONVIFCamera('10.16.0.147', '80', 'admin', '123456', '/etc/onvif/wsdl                                                                                                                                                             /')
    # Create media service object
    media = mycam.create_media_service()
    # Create ptz service object
    ptz = mycam.create_ptz_service()

    # Get target profile
    media_profile = media.GetProfiles()[0]

    # Get PTZ configuration options for getting continuous move range
    request = ptz.create_type('GetConfigurationOptions')
    request.ConfigurationToken = media_profile.PTZConfiguration._token
    ptz_configuration_options = ptz.GetConfigurationOptions(request)
    request = ptz.create_type('ContinuousMove')
    request.ProfileToken = media_profile._token

    ptz.Stop({'ProfileToken': media_profile._token})

    # Get range of pan and tilt
    # NOTE: X and Y are velocity vector
    global ZMAX, ZMIN
    ZMAX = ptz_configuration_options.Spaces.ContinuousZoomVelocitySpace[0].XRang                                                                                                                                                             e.Max
    ZMIN = ptz_configuration_options.Spaces.ContinuousZoomVelocitySpace[0].XRang                                                                                                                                                             e.Min

#PT control

    zoom_out(ptz, request)

if __name__ == '__main__':
    continuous_move()
sysadmin@1001-noc-nvr-01:/etc/motioneye$
`

starbasessd, 2021-11-08:
Don't forget the rest of my questions...

starbasessd, 2021-11-08:
An add: What OS are you doing this in?

doucettom, 2021-11-08:
Pan/tilt/Zoom are available in the Camera's web UI but they're javascript buttons.
This is the camera:
https://www.amazon.com/Hikvision-Compatible-18xOptical-Security-4818X-IZ/dp/B089M9WR7L
I'm using Ubuntu Server 20.04.3

doucettom, 2021-11-10:
[image]
This is what the PTZ controls of the Cam web ui are looking like.

[image]
This is the PTZ configuration menu.

JamBalaya56562, 2026-11-10:
Closing, as this is about the script itself: motionEye's zoom buttons only run `/etc/motioneye/zoom_in_1` and `zoom_out_1` for camera 1, so what they do is up to the ONVIF calls in them, and the camera's zoom support there. Please open a new issue if the buttons themselves do not run the scripts.

closed 2026-11-10 by JamBalaya56562