motioneye-project/motioneye #2226: Can I use python 3.7?

johncblacker, 2021-11-10:
The default release of my rPi 3b+ is python 3.7.3.  Can I use it to run motioneye or do I need to use python 2.7 (which I have available but don't use).  I tried on Python 3.7.3 but ran into a problem with the systemctl start motioneye command not being able to start the server: "from motioneye.meyectl import main" getting ModuleNotFoundError: No module named 'motioneye' but it's threre if I do a pip show motioneye.

starbasessd, 2021-11-10:
Currently (as of 10 Nov 2021) Python2.7 is required.

starbasessd, 2021-11-10:
Please use the install instructions from the WiKi, depending on if you are using Buster or Bullseye.
https://github.com/ccrisan/motioneye/wiki/Installation

johncblacker, 2021-11-11:
Tried installing under python2.7 but ran into issue with pycurl not being installed and when I tried to install pycurl I ran into problems, here's the output of attempt to install pycurl
--------------------------------------------------------------------------------------------------------
pi@rpi3:~ $ pip2 install pycurl
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pycurl
  Using cached https://files.pythonhosted.org/packages/ef/05/4b773f74f830a90a326b06f9b24e65506302ab049e825a3c0b60b1a6e26a/pycurl-7.43.0.5.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-32ctPG/pycurl/setup.py", line 961, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-install-32ctPG/pycurl/setup.py", line 623, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-install-32ctPG/pycurl/setup.py", line 101, in __init__
        self.configure()
      File "/tmp/pip-install-32ctPG/pycurl/setup.py", line 238, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-32ctPG/pycurl/
----------------------------------------------------------------------- end of log output --------------------------------------
Not sure how to proceed at this point.

starbasessd, 2021-11-11:
Off the top, wasn't run as admin (sudo).
What base OS? Which instruction set?

johncblacker, 2021-11-11:
No, I didn't run as sudo pip2 as most posts on internet caution against doing so.  But, I got around that problem by running sudo apt-get install python-pycurl.  But, now I get same import error when I try to bring up server.  So, I'm wondering whether I have to start completely from the beginning - including removing and reinstalling motioneye and motion again?  When I installed motioneye using pip2 install motioneye, then looked at the pip2 list output, all the pre-reqs were there as was motioneye, but the systemctl start motioneye still fails with ImporError: No module named motioneye.meyectl.  I started python2.7 from command line, then did import motioneye.meyectl and it worked just fine.  help(motioneye.meyectl) indicated that the file is located at '/home/pi/.local/lib/python2.7/site-packages/motioneye/meyectl.py'....I'm wondering where it's looking when running as a service?

johncblacker, 2021-11-11:
So!  I did: sudo python2.7 then import motioneye.meyectl and it fails with import error.  When systemctl start motioneye is issued, the command will be running under root, unless told otherwise...

starbasessd, 2021-11-11:
Correct. MotionEye on another OS assumes it is installed and running as the root user.
There are instructions at https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User
to set up and run as a regular user. It is no where near as stable as the regular install.
You didn't answer the other 2 questions, which would help me help you.

johncblacker, 2021-11-11:
OK, here's what you requested:
Linux rpi3 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 arm7l GNU/Linux.

johncblacker, 2021-11-11:
One way to fix the problem is to add the following to the [Service] section of "motioneye.service":
Environment="PYTHONPATH=/home/pi/.local/lib/python27/site-packages"

needed because pip2 installed motioneye into that directory; whereas the packages installed via dpkg or its derivatives will install into dist-packages.  I chose not to run as a local user because you mentioned that isn't as reliable nor stable as running under root.  Seems like this is the best situation that avoids a lot of potential issues; yet this is likely a universal solution, I believe.  I also think, though, that had I installed motioneye in python by first setting up a virtualenv it'd probably have worked out of the box...

starbasessd, 2021-11-11:
Restart the install process with sudo pip2 install motioneye. This will make it available to all users. After re-completing the rest of the steps, follow the instructions as "Running as a Normal User". You will then have motionEye running as a non-root user. You can _try_ to run pip2 as the 'normal' user, but it is not tested...

johncblacker, 2021-11-11:
I'm going to close this issue; you can decide how/if ever you want to handle it if it comes up again.  My solution of adding the environment spec inside the service file works and I've got the camera working now.  I already have a script for sending notifications via pushsafer and that works fine; now I've got to include a still picture triggered by motion along with the notification.  Thanks for your help.  Since my resultant issue doesn't seem to be python version related, I wonder if it'd work with python 3...I might try it.

closed 2021-11-11 by johncblacker