motioneye-project/motioneye #2227: Fit Frames Horizontally

Gonioul, 2021-11-11:
Hi,

Motioneye can't shrink videos to fit width (accounting layout columns), like it can do for height with "Fit Frames Vertically", it would be nice to have a similar option for height, for small screen like tablet or phone, or global preview for many cameras.

Regards,

MichaIng, 2022-03-19:
This is done automatically, isn't it? On large screens, the chosen number of horizontal frames is matched to the browser width. If the browser width is less than 1200px, a single frame is shown per line (instead of the configured number, GUI setting hidden as well) and matched to the width.

On my tests, the vertical fit on the other hand has no effect, but the aspect ratio is instead kept at all times 🤔.

zagrim, 2022-03-19:
@MichaIng see my comment on PR #1570 regarding vertical fit. For horizontal fit my testing results agree, does not seem to work.

MichaIng, 2022-03-19:
Now I see it, when I reduce the height of the browser window much, or switch to 1-2 cameras or portrait cameras, "Fit Frames Vertically" reduces the sizes to have one row fully shown in the browser window, which is otherwise not the case.

"Layout Rows" however seems to not have the intended effect: When I select 2, I'd expect the size to be reduced so that always two rows are fully shown, but it remains so that always only one row is matched the window height. I see the `width` frame element attribute being set by JavaScript, should be easy to find out what's going on.

> For horizontal fit my testing results agree, does not seem to work.

Does **not**? In my case, the frame width is always reduced so that the number of defined "Layout Columns" us always fully shown in the browser window, regardless how any other settings are set. When the window width is below 1200px, a single frame is however enforced.

So basically by default camera frames are shrunk so that always "Layout Columns" (or one with <=1200px window size) fits into the window, and with "Fit Frames Vertically" they are additionally shrunk so that ways 1 row fits into the window, with "Layout Rows" now having any effect I could recognise so far.

zagrim, 2022-03-20:
>  >For horizontal fit my testing results agree, does not seem to work.
> 
> Does not? 

Sorry, got confused there somehow. Yes, the camera frames do always fit the available space horizontally.

JamBalaya56562, 2026-11-13:
As @MichaIng and @zagrim found above, the frames already fit the window width: with *Layout Columns* set to 2, two frames share it (checked on the current dev: two 472 px frames side by side in a 1024 px window). The other part of the discussion, *Layout Rows* being ignored by *Fit Frames Vertically*, was fixed in 0.45.0b1 by #3485.

closed 2026-11-13 by JamBalaya56562