motioneye-project/motioneye #2228: Alerts with latest motion version (4.4.0)

JC38, 2021-11-15:
Hello,

I've updated my motion version to the latest 4.4.0 and I can see some red alerts about settings. Everything seems to work but I would like to supress these alerts.
Thanks for your help

> -- Logs begin at Sun 2021-11-14 11:50:18 CET, end at Mon 2021-11-15 09:05:41 CET. --
nov. 14 11:50:19 motioneye systemd[1]: Started motionEye Server.
nov. 14 11:50:24 motioneye meyectl[448]:     INFO: hello! this is motionEye server 0.42.1
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [ALR] [ALL] conf_cmdparse: "netcam_keepalive" replaced with "netcam_params" after version 4.3.2
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [ALR] [ALL] conf_cmdparse: "netcam_tolerant_check" replaced with "netcam_params" after version 4.3.2
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [ALR] [ALL] conf_cmdparse: "netcam_keepalive" replaced with "netcam_params" after version 4.3.2
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [ALR] [ALL] conf_cmdparse: "netcam_tolerant_check" replaced with "netcam_params" after version 4.3.2
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-3.conf
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [ALR] [ALL] conf_cmdparse: "netcam_keepalive" replaced with "netcam_params" after version 4.3.2
....
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-12.conf
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [ALR] [ALL] conf_cmdparse: "netcam_keepalive" replaced with "netcam_params" after version 4.3.2
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [ALR] [ALL] conf_cmdparse: "netcam_tolerant_check" replaced with "netcam_params" after version 4.3.2
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.4.0 Started
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
nov. 14 11:50:27 motioneye motion[576]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
nov. 14 11:50:29 motioneye meyectl[448]:    ERROR: could not find motion camera id for camera with id 13
nov. 14 11:50:29 motioneye meyectl[448]:     INFO: cleanup started
nov. 14 11:50:29 motioneye meyectl[448]:     INFO: wsswitch started
nov. 14 11:50:29 motioneye meyectl[448]:     INFO: tasks started
nov. 14 11:50:29 motioneye meyectl[448]:     INFO: mjpg client garbage collector started
nov. 14 11:50:29 motioneye meyectl[448]:     INFO: server started

starbasessd, 2021-11-15:
These will require a update modification to motioneye. I will flag as 'bug', and 'enhancement' for a future revision. 4.3.2 doesn't produce these messages. What feature/functionality does 4.4.0 have for your use case?

starbasessd, 2021-11-20:
More camera issues explained by RPiFoundation:
https://www.raspberrypi.com/news/bullseye-camera-system/#comment-1566990

zagrim, 2023-12-07:
Since #2462 has already been merged this issue should've been fixed long time ago, so closing this. The PR mentions one issue with changed Motion default behaviour in some case with MJPEG netcams (if I understood correctly) but I don't think that should be a reason to keep also this one open.