motioneye-project/motioneye #2229: not recording video stream

githubsharezone, 2021-11-17:
i install the software on debian 9 but whne the system trigger event not recording on folder. 
i see log and show

1:ml1:Stanzetta] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Invalid argument
[1:ml1:Stanzetta] [ERR] [NET] ffmpeg_open: Failed to allocate codec!
[1:ml1:Stanzetta] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/etc/motioneye/dvr/stanzetta/2021-11-17/12-14-49.mp4]
[2:ml2:Sala] [ERR] [ENC] ffmpeg_set_codec: Could not open codec Invalid argument
[2:ml2:Sala] [ERR] [NET] ffmpeg_open: Failed to allocate codec!
[2:ml2:Sala] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/etc/motioneye/dvr/sala/2021-11-17/12-14-49.mp4]

root@armhdvr:~# uname -a
Linux armhdvr 4.19.0-18-arm64 #1 SMP Debian 4.19.208-1 (2021-09-29) aarch64 GNU/Linux

githubsharezone, 2021-11-19:
i resolved change the type codec from gui

closed 2021-11-19 by githubsharezone

common-guid, 2022-03-06:
@githubsharezone - can you explain how you were able to change the codec from the gui? I am having the same issue, but do not see a codec option.

starbasessd, 2022-03-06:
Settings, select correct camera, Movies, Movie Pasthrough = OFF, Movie Format (Select)
I do not use any OMX formats on non-Pi hardware.