motioneye-project/motioneye #2230: esphome - esp32_camera_web_server component does not work with motioneye

sztanpet, 2021-11-17:
The report that advised me to make an upstream issue was https://github.com/hassio-addons/addon-motioneye/issues/325

Trying to get my espcam cameras to work with motioneye, which with the new esphome release should be possible thanks to the https://esphome.io/components/esp32_camera_web_server.html component.
I successfully managed to make it work (thanks to help from jesserockz) with frigate, where the configuration needed was:

```yaml
cameras:
  espcam1:
    ffmpeg:
      inputs:
        - path: http://espcam1.local:8080
          roles:
            - detect
      input_args: -avoid_negative_ts make_zero -fflags nobuffer -flags low_delay -strict experimental -fflags +genpts+discardcorrupt -use_wallclock_as_timestamps 1
      output_args:
        record: -f segment -segment_time 10 -segment_format mp4 -reset_timestamps 1 -strftime 1 -c:v libx264 -an
        rtmp: -c:v libx264 -an -f flv
```

(I disabled frigate when trying to configure motioneye, I did read the esp32 camera webserver only supports a single stream at a time)

The error message says "request timed out" for both network camera and simple mjpeg camera options.
The debug log contains the following:

```
[15:22:25] INFO: Successfully send discovery information to Home Assistant.
   DEBUG: loading additional config structure for main, with separators
   DEBUG: additional config section: network
   DEBUG: loading additional config structure for camera, with separators
/bin/sh: lsb_release: not found
   DEBUG: 304 GET / (127.0.0.1) 122.44ms
   DEBUG: 200 GET /login/?_=1637158947536&_username=admin&_signature=a75453c2fb107661a47c9c7e13b1610e824ed130 (127.0.0.1) 1.12ms
   DEBUG: preferences file "/data/motioneye/prefs.json" does not exist, using default preferences
   DEBUG: 200 GET /prefs/?_=1637158947570&_username=admin&_signature=90cd650b83c33652ba3be4aa19bc17c854f7c711 (127.0.0.1) 0.92ms
   DEBUG: getting main config
   DEBUG: 200 GET /config/main/get/?_=1637158947705&_username=admin&_signature=ae7289d42359162958d408d45a6769478bfe2857 (127.0.0.1) 0.95ms
   DEBUG: listing cameras
   DEBUG: 200 GET /config/list/?_=1637158947750&_username=admin&_signature=bb83b874ed5421f77b362ee37b0d52ff7998f7a3 (127.0.0.1) 1.02ms
   DEBUG: listing cameras
   DEBUG: listing V4L2 devices
Cannot open device /dev/video0, exiting.
   DEBUG: 200 GET /config/list/?_=1637158949165&username=&password=&proto=v4l2&_username=admin&_signature=78c717dd81b9998a299fdf0b9cb817ef43e144a7 (127.0.0.1) 9.35ms
   DEBUG: listing cameras
   DEBUG: testing (m)jpg netcam at http://espcam1.local:8080/ using no authentication
   DEBUG: 200 GET /config/list/?_=1637158960503&scheme=http&host=espcam1.local&port=8080&path=/&username=&password=&proto=mjpeg&_username=admin&_signature=b455566d9be0ab0db8a4d7abb67b28ba6e63652f (127.0.0.1) 10002.88ms
   DEBUG: listing cameras
   DEBUG: testing (m)jpg netcam at http://espcam1.local:8080/ using no authentication
   DEBUG: 200 GET /config/list/?_=1637158975206&scheme=http&host=espcam1.local&port=8080&path=/&username=&password=&proto=netcam&_username=admin&_signature=761042a0c4aa7dc9bd6364217980afc2cbc6364b (127.0.0.1) 10003.15ms
```

the raw output for a single "frame" of the stream from the esp is as follows:

```
$ curl -v http://192.168.1.49:8080/
*   Trying 192.168.1.49:8080...
* Connected to 192.168.1.49 (192.168.1.49) port 8080 (#0)
> GET / HTTP/1.1
> Host: 192.168.1.49:8080
> User-Agent: curl/7.80.0
> Accept: */*
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 200
< Access-Control-Allow-Origin: *
< Content-Type: multipart/x-mixed-replace;boundary=123456789000000000000987654321
* no chunk, no close, no size. Assume close to signal end
<
--123456789000000000000987654321
Content-Type: image/jpeg
Content-Length: 14054

BINARY CONTENT SKIPPED
```

starbasessd, 2021-11-17:
Can you remove HA from the mix? The error involves HA.
What base OS are you using?
In order to best diagnose the issue, we need to simplify what we are working with.
You don't need to wipe your current setup, just use another SDCard or USB drive.
Install a fresh copy of RaspberryPiOS or whatever base OS you are using.
Follow the appropriate install instructions from:
https://github.com/ccrisan/motioneye/wiki/Installation
Then try to connect motionEye to your ESP32_web_camera
Does your ESP support mjpeg, mp4, avi, rtsp, rtmp, h.264 video stream?
There are a number of suggestions to connect to ESP32 type cameras depending on what software you are running on them here:
https://www.ispyconnect.com/camera/esp32
They have an excellent database, and there are only a couple out of the thousands overall listed that motion/motionEye/motionEyeOS can't work with...
If you respond, please reply to ALL the questions asked.

sztanpet, 2021-11-17:
> Can you remove HA from the mix? The error involves HA.

sure

> What base OS are you using?

HASS-OS in this instance, motionEye is run in a docker container, see the link in my top post for the Dockerfile if interested.

> Then try to connect motionEye to your ESP32_web_camera

I have now installed motionEye on my arch box and tried to connect with it to my esp32cam too, the exact same problem.
"request timed out", tried using the IP of the esp too, same issue.

curl works, its output is the same as in the top post
ffplay also works, opening the url in the browser also works

> Does your ESP support mjpeg, mp4, avi, rtsp, rtmp, h.264 video stream?

it provides the simplest mjpeg stream, see the curl output for an example, any other URL apart from `http://192.168.1.49:8080/` returns a 404

starbasessd, 2021-11-17:
Did you try: (with both Network Camera & Simple MJPEG Camera)
http://192.168.1.49:8080/?action=stream
http://192.168.1.49:8080/stream
http://192.168.1.49:8080/Stream
I don't know, I do not have one here to play with. I have a number of cameras that all connect to motionEye and motionEyeOS in docker, native, and modified test beds. The only time I have issues is using a certain setup involving HA, but I get accused of 'pointing fingers'. I asked you to do a clean install using the instructions from the motionEye wiki to see if it works by itself, but didn't see that result. Installing using their docker install is not the same thing.

cap9qd, 2021-11-17:
I have a camera already setup and updated ESPhome and updated my camera
YAML to include the web server component.

I believe the issue is the "add camera" dialog box forces a trailing "/"
and the esp32 web server doesn't handle this properly.

I am going to do some more testing tonight when I have more time.

starbasessd, 2021-11-17:
@cap9qd 
Thank you for helping. If you go into the CameraX.conf file, can you remove the trailing '/' and does that fix the issue?

cap9qd, 2021-11-17:
I will have to build the conf file since it won't get past the add camera
dialog with the timeout error.

cap9qd, 2021-11-17:
I built the conf file from a different camera and didn't have any luck. I
may be off in the weeds here but I'll mess with this more tonight. Having
esphome and an esp32 camera is something I'm interested in getting working
with MEyeOs.

cap9qd, 2021-11-17:
Okay so I cant just let something go...

I managed to get it working and the trailing "/" was just just a wag and
didn't have anything to do with it in the end.

I was able to get it to work by defining it as a stream type mjpeg in the
url in the motion conf cile for the camera. It still doesn't work in the
"add camera" dialog box with the url.

The url I used on the conf file is mjpeg://192.168.0.abc:8081/ (my
streaming port is defined in the device's esphome yaml is set to 8081) in
the camera-N.conf I made. Idk how the auto discovery stuff works in
Motioneye for the dialog but it seems it doesn't like whatever the camera
reports back; I managed to do something similar with an ESP32 camera in the
past but bailed on the project due to my terribly buggy code.

I also made sure the height, width and frame rate were consistent between
the esp camera config and motion camera configuration file.

sztanpet, 2021-11-18:
So I guess the question is what does motioneye expects to see on the URL entry screen with regards to the returned response?

cap9qd, 2021-11-18:
It does not appear to work properly from the add camera dialog box. I used
the stream type of mjpeg:// in my created conf file. You could do the same
to make it work by copying an existing file and adding it to the
motion.conf file.

I am not well versed in the MotionEye code base so I will take a look but I
hope someone more familiar with it can take a look.

I tried several ways to 'trick' the dialog but no results. It may have
something to do with what the esp returns.

ayufan, 2021-11-21:
https://github.com/esphome/esphome/pull/2777

starbasessd, 2021-11-21:
Thank you, @ayufan

sztanpet, 2021-11-23:
So I guess this can be closed because it's being fixed on the esphome side

closed 2021-11-23 by sztanpet