motioneye-project/motioneye #2231: Pi4 Image File does not work

myxxmikeyxx, 2021-11-18:
The newer Pi 4's say start4.elf does not work with the new hardware. 
Then it has the link to the raspberry pi website and says download the latest image.

starbasessd, 2021-11-18:
You need to update your Pi4 eeprom to something from this year using RaspberrPiOS.
You can check your RPi4 eeprom with 
vcgencmd bootloader_version
The date it shows must be after 1 Jan 2021. If not, you can do a 
sudo rpi-eeprom-update -a
to get the current version

starbasessd, 2021-11-18:
BTW, you need motioneyeos-raspberrypi4-dev20201026 
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz
for the RaspberryPi4. Previous versions had major issues, especially with RPi4-4GB & 8GB
This is the motionEye github, for motionEye on other OSs. You want
https://github.com/ccrisan/motioneyeos/

myxxmikeyxx, 2021-11-18:
Thank you very much I will give it a try, and my apologies I did not realize it had a separate repo

myxxmikeyxx, 2021-11-18:
vcgencmd bootloader_version returned `Apr 29 2021` 
So it is up to date. I think I might have been using the wrong img.
I had version `motioneyeos-raspberrypi4-20200606.img.xz`
I have installed the img you linked and it is booting right now.

closed 2021-11-18 by myxxmikeyxx

starbasessd, 2021-11-18:
Report if it comes up all the way, please.

myxxmikeyxx, 2021-11-18:
It did, the website loaded!

Side note: I don't think it will work since I read the wiki for a moment and it says it is made from `buildroot`.
I just know buildroot doesn't have a package manager because it is made to be super light weight, so I don't know how to install vlc unless I try to build my own.
I want to be able to use the on motion detection run Script.sh and use vlc to play an mp3 file.

myxxmikeyxx, 2021-11-18:
@starbasessd Thank you for all the help!

starbasessd, 2021-11-18:
True, you can't load packages like VLC. I use Debian lite or Ubuntu lite on a M73 running motionEye and use mpg123 to play a doorbell when it sees someone walking up to a door or in the driveway. VLC is very overkill for that use case...

myxxmikeyxx, 2021-11-18:
Thank you, I had just done a quick google search and it was one that looks to be able to play a sound easily.
I will have to check out `mpg123`.