motioneye-project/motioneye #2232: Pull Request for Wiki

myxxmikeyxx, 2021-11-18:
I have updated the wiki some. I changed some download links, document links, and removed duplicate documentation.
You will have to pull from my [fork wiki here](https://github.com/myxxmikeyxx/motioneye/wiki)
I changed the Installation.md to a new raspberry pi link
Made a `Install-On-Raspberry-Pi.md`
Added links to the different raspberry Pi installs.

You could just change the side bar and add another clickable links in the installtion.md but I figured in the future it might be easier to have a main link one and just link to the newer documentation when its created.

starbasessd, 2021-11-18:
Please make a zip of the files you changed with notes as to what you changed and why, vs what changes I've made recently  and attach here.

myxxmikeyxx, 2021-11-18:
[motioneye.wiki.zip](https://github.com/ccrisan/motioneye/files/7565452/motioneye.wiki.zip)
`Installation.md` changed raspberry pi link from `Install-On-Raspbian-Buster.md` to `Install-On-Raspberry-Pi.md`
`Install-On-Raspberry-Pi.md` Created file, Just a place to put tips and update links in one location
`Install-On-Raspbian-Bullseye.md`
`Install-On-Raspbian-Buster-or-earlier.md` Just removed the duplicate document that was the exactly the same
`Install-On-Raspbian-Buster.md` Removed file
`Install-On-Raspbian.md` Removed file
`_Sidebar.md` Changed the order and links

myxxmikeyxx, 2021-11-18:
Also thanks for the help on how to properly submit it. I was trying to follow the documents but the links took me to buster at first but the side bar had the link so I used the side bar and after cloning the wiki I saw the old unused docs. I wish you could just make a pull request for the wiki, maybe someday github will let you

myxxmikeyxx, 2021-11-19:
Realized today that I closed this on accident.

JamBalaya56562, 2026-10-28:
Thanks for the proposal. The per-distribution installation pages have been removed from the wiki (December 2025), including the pages this proposal changes. There is now a single install method for all distributions in the README: https://github.com/motioneye-project/motioneye#installation. Closing, as the pages this refers to no longer exist. If the README instructions do not work for you, please open a new issue.

closed 2026-10-28 by JamBalaya56562