motioneye-project/motioneye #2233: Motioneye SSH Sudo Commands Not Working?

IsmaelPR1977, 2021-11-19:
I have the latest version of Motioneye on my Raspi 3 B+ with 2 usb cameras and all is working well. All I want to do is swap the order that they appear in but no matter what I try in SSH (Putty) I get a **-sh: sudo command not found**. What exactly is going on? I just want to rename the camera-1.conf and camera-2.conf files (swap them) through Putty, is that not possible?

Thanks!
Iz

starbasessd, 2021-11-19:
Running motionEye or motionEyeOS?
If motionEyeOS, you are already admin/root, no need to use sudo.
If you want to modify conf files, mod the ones in /data/etc (most of the OS is read-only)

IsmaelPR1977, 2021-11-19:
I am running MotioneyeOS. So I can just use the commands without sudo?

starbasessd, 2021-11-19:
Correct.

IsmaelPR1977, 2021-11-19:
Thank you for clearing that up so quickly! All set now.

closed 2021-11-19 by starbasessd