motioneye-project/motioneye #2234: Another CPU usage question - motioneye vs motioneyeos

rocket59, 2021-11-19:
I recently got a Wyze Cam 3 flashed with RTSP firmware. On a raspberry pi 3B with the latest motioneye (manually installed) CPU usage hit 100% with no detection options. However when I used motioneyeos instead (same pi same camera) CPU usage dropped to about 50%. What's the difference between the 2?

starbasessd, 2021-11-19:
motionEyeOS is a VERY stripped down debian/raspbian (buildroot/thingOS) that only has enough stuff installed to run motion, motionEye, and support functions (mostly written in python2.7). If you are running motionEye on another OS (even Raspbian lite or debian lite) there is a lot of other stuff running for OS support functionality to add other features. There may be an extra copy of motion running in motionEye on another OS, also if you didn't follow the latest Debian or Raspbian OS installs. There may be a different version of motion running, also. Lots of things could be affecting it...

rocket59, 2021-11-19:
Actually that explains why I see a few instances of motion in htop when using motioneye. Any way to disable these?

starbasessd, 2021-11-19:
It's in the current documentation, depending on OS. In Debian & RPiOS you can do:
sudo systemctl stop motion
sudo systemctl disable motion
sudo reboot
motionEye starts any instances of motion it needs.

rocket59, 2021-11-19:
Ok thanks.

closed 2021-11-19 by rocket59