motioneye-project/motioneye #2235: MotioneyeOS on Pi0 2 w

JBoyKey, 2021-11-20:
Have loaded Motioneye on hassio and Pi0, but am unable to get my new Pio 2 w to boot, loaded latest MotionEyeOs from Supported Device page - Raspberry PI (A, B, A+, B+, Compute Module, Zero and Zero W models) burned it with BalendaEctor and nothing, it boots to splash screen and sits there. Created Supplicant.conf just like other working Pi0 but don't see it on router, sorry no logs, thanks in advance Joe.

starbasessd, 2021-11-20:
There are options in the motionEyeOS github:
https://github.com/ccrisan/motioneyeos/issues
It is not supported at this time, but should be in the next official release.

closed 2021-11-20 by starbasessd

JBoyKey, 2021-11-20:
Cool, THANKS, searched everywhere to no avail, awesome stuff - motionEye, now upto 5 Pi's, joe.

starbasessd, 2021-11-20:
It is supported if you load another OS (Raspbian or Debian) from the install page here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye
https://github.com/ccrisan/motioneye/wiki/Install-On-Debian
https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye

JBoyKey, 2021-11-20:
That’s what I’m doing on a Pi4b8g with BullsEye, but stuck getting it to view it in Browser, keep getting this text with buttons and pointer arrow icons, will send screenshot when aI can, thanks Joe

 

https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye
https://github.com/ccrisan/motioneye/wiki/Install-On-Debian
https://github.com/ccrisan/motioneye/wiki/Install-on-Raspbian-Bullseye