motioneye-project/motioneye #2236: MotionEye not sending commands on detection after a while

StevenSlaa, 2021-11-22:
Hello everyone,

I have an issue.
I'm running motioneye from my Home Assistant server.
I have one camera (for now) that should run a command on detection.
The command for this is the following:
```
mosquitto_pub -h 127.0.0.1 -t homeassistant/cameras/parkeerplaats/motion -m "true" -q 2
```
This will set an MQTT property to `true` with `QoS` 2

In Home Assistant this topic is converted into a motion sensor, which turns my outdoor lights on.

To return the property back to `false`, I send the following command:
```
mosquitto_pub -h 127.0.0.1 -t homeassistant/cameras/parkeerplaats/motion -m "false" -q 2
```
This will set an MQTT property to `false` with `QoS` 2
I send this command at `Run An End Command` and also(because the previous command doesn't always trigger) when file is saved to my NAS (Samba share).

**The issue:**
This system works for a while but at some point it stops sending the commands. The detection is working, but It does not update the MQTT topic anymore.

I jumped in front of the camera 15 times and everything works great! But 30 minutes later when I try it again it doesn't work anymore. When looking at the MQTT topic it doesn't change either. Which traces the issue back to sending the commands. This because it is still detecting me perfectly fine.

In the logs I can't seem to find anything related to this issue.

My best gues is that it could be something related to samba thats making it not sending commands. Does anyone know the problem, or a way to figure out what's going on?😉👍

Regards,

Steven

starbasessd, 2021-11-22:
All my assistance in these cases is to take Home Assistant out of the equation first. If you are using a docker of mqtt and motionEye, this is easy enough. 
If you are unable or unwilling to do this, I cannot help you as HA creates a number of issues by itself.
I will leave it open in either case to see if anyone else can assist, too.

JamBalaya56562, 2026-11-10:
Closing, as this got no follow-up and was with the Home Assistant add-on of 2021. Please open a new issue with `motion.log` and `motioneye.log` from the time the commands stop if this still happens with a current release.

closed 2026-11-10 by JamBalaya56562