motioneye-project/motioneye #2237: Change default port for internal streaming from 8080 + camera_id to something less likely to overlap

surdaft, 2021-11-24:
**TL;DR**
- Port 8080 and 8081 are extremely common for applications hosting a http server on non-privilege ports
- Changing from 8081 to 8181 would prevent a few people not using docker from encountering a similar issue
- It shouldn't affect anybodies experience as its just the default and can be overriden / defined

**Issue**

I was diagnosing an issue on my motioneye setup. I am not using the docker container, as I wanted to save on resources on the PC, so I have motioneye set up directly on the machine with nginx. It is shared with one or two other programs, like supervisor.

I noticed that in syslog that it said "mjpg client timed out receiving data for camera 1 on port 8081" - so referencing the camera-1.conf file that was the stream_port, used to send out the camera data from motion which motioneye then picks up and displays to the user. So it makes sense why it wasn't displaying, if it couldn't get anything on port 8081.

I initially had supervisor running its inet interface on this port. However I am raising this issue to then create a PR to change it to a port less likely to overlap with other services. As 8080 and 8081 are extremely common for any kind of http related project binding to a non-privilege port.

https://github.com/ccrisan/motioneye/blob/289b23a81548a6410a91e54bd224ce94e8e6cd75/motioneye/config.py#L1936

starbasessd, 2021-11-24:
Could just as easily blame other apps for using the same ports, not just motionEye/motionEyeOS. Those ports are used because everyone knows them as alternatives. Why not pick 18080 or 28080, 38080 or whatever?

surdaft, 2021-11-24:
I'm just updating the default so it's less likely to overlap with other items. It shouldn't be an issue for a user to modify the config because they had something else installed on their box, if you could just not use such a common port.

We could totally go for 18080 if you like, I just used 8180 as a quick solution.

starbasessd, 2021-11-24:
No, I'm voting no change for default, for the reason you stated: "It shouldn't be an issue for a user to modify the config because they had something else installed on their box". 
I deal daily helping with people who have no clue even where to begin to make those mods. I also deal with users who use apps like HomeAssistant and others that depend on motionEye to be at certain ports. Plus the users who would create their system using one version, then add cameras using another version. I can imagine the headaches that will cause.

surdaft, 2021-11-24:
Fair enough.

closed 2021-11-24 by surdaft