motioneye-project/motioneye #2240: Don't record motion when camera is in Home security mode

mjf2708, 2021-11-29:
I am using MotionEye on a Raspberry Pi to record video in response to motion triggers on two Eufy cams: Indoor 2K Cam and Indoor 2K Cam Pan & Tilt.

Within the Eufy security app, I am using geofencing to switch between Home and Away security modes, and I have configured Home mode to not record video or generate push notifications. (Away mode is configured to do both these.)

However, MotionEye still generates videos whenever it detects motion on these cams when they are in Home security mode. Is there any way to prevent this?

starbasessd, 2021-11-29:
You could try enabling motion web interface and telling it to pause detection and enable detection.
https://motion-project.github.io/motion_config.html#OptDetail_Webcontrol
curl {IP}:{port}/{camid}/detection/start Start or resume motion detection.
curl {IP}:{port}/{camid}/detection/pause Pause the motion detection.
If you have multiple cameras you can bash script all the cameras into a single script to pause and another to enable.
so, for example you want to pause camera1:
curl localhost:7999/1/detection/pause
and to restart camera1:
curl localhost:7999/1/detection/start
(assuming the Eufy app and motioneye/motion are on the same machine. and not in a docker container.)

starbasessd, 2021-11-29:
If there is no way for the Eufy app or cameras to indicate or call the curl command, then no. motionEye can accept GPIO triggers, software triggers, etc. There are other issues posted where people have written scripts and software to watch for MAC addresses on the local network, but I don't think anyone has written anything to geofence the way some apps do (like my Thermostat app)...

mjf2708, 2021-11-29:
OK, for me the solution is to use Disarmed mode in the Eufy app, as even when Record Video and Push Notification are turned off in Home mode in the Eufy app, MotionEye still responds to motion triggers (suggesting some strange behaviour in the Eufy app?).

I can then use geofencing to switch between Away and Disarmed.

starbasessd, 2021-11-29:
If your phone auto connects to the local network, you can write a script using nmap to watch for a MAC address on the network (your phone or other device) with something like:
sudo nmap -sn 192.168.2.* | grep 00:11:22:33:44:55 (substitute your mac address...)

mjf2708, 2021-11-29:
I've successfully used the curl commands to pause / start detection, and so will incorporate these into a script, as recommended.

Thanks.

closed 2021-11-29 by mjf2708