motioneye-project/motioneye #2241: on_picture_save action and on_movie_end not separated

Maja61, 2021-12-01:
Hi,

i try to test the configuration of motioneye and i want to have a possibility to set different actions for "picture save" and "movie end". At the end i want to sent the oic when it is saved and i want to send the movie when the movie is save, both via telegram with  seprated shell scripts. 

In the GUI Setup i found the option "File Storage --> Run a command". But tis option effects "on_picture_save " and "on_movie_end " at the same tiime with the same command. 

[image]
[image]

Is it possible to configure both separatly?

Thanks & best regards

My setup:
motionEye Version | 0.42.1
Motion Version | 4.4.0
OS Version | Raspbian 11 on PI4

starbasessd, 2021-12-01:
There is also in Motion Notifications a Run End Command when enabled gives you:
[image]
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t; test_command
Does this have the effect you want?
Or, once created, you could edit the lines and change to what you wanted for each...

cap9qd, 2021-12-01:
I wrote a BASH script to only trigger the command if the extension is jpg
tog et around this "issue".

pFile=%f; if [ "${pFile: -4}" == ".jpg" ]; then ##command(s)##; fi

Maja61, 2021-12-09:
> There is also in Motion Notifications a Run End Command when enabled gives you: [image] on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh [...]

No, that doesnt help. my target is to send the last pic with %f  "on_picture_save" with additional script and to send the last video with %f "on_movie_end " with additional script. 

If i change it in the camera.conf it will be overwritten everytime when i change something in the gui afterwards.  

Why does the command from the "File Storage" effects "on_picture_save" and "on_movie_end " at the same time? For my opinion it should be separated listed in the GUI or only one should be affected by the adjustment via GUI and the other one can be set via "Extra Motion Options" . 

Is it possible to change it?

starbasessd, 2021-12-09:
Not currently. I will tag as 'enhancement' and 'feature request' to separate the settings.

Maja61, 2021-12-09:
Can i do a small adjustment in my installation to get it separatly as an workaround until the enhancement will be implemented?

starbasessd, 2021-12-09:
If you are a python 2.7 programmer, sure. The code is all there. I am not, I just do support. I think it would be a little more involved than just changing a line or 2, which is why I tagged for enhancement/feature.

Maja61, 2021-12-10:
ok, thx then i will wait for the new release.

starbasessd, 2021-12-10:
No guarantees...

Maja61, 2021-12-10:
> pFile=%f; if [ "${pFile: -4}" == ".jpg" ]; then ##command(s)##; fi

@cap9qd can you explain it how i can do it in your way? Where i have to paste your line exactly?

cap9qd, 2021-12-10:
I put it in the box under "File Storage" in the box under the checkbox for
"Run A Command". This little "script" only runs my command if it's a
picture file. You could do the same with a python or bash script. I did
this so I didn't have any scripts to restore if my system fell down. You
could easily test this offline in a BASH script and add a second if
statement for the video file.

Maja61, 2021-12-13:
@cap9qd now i got it, thx workaround is working :-)