motioneye-project/motioneye #2242: Problem installing motioneye

nikscha, 2021-12-01:
Noob here, please be patient with me lol
I am trying to install motioneye+motion on a raspi 4B running Raspian.
I am following the steps in the wiki almost exactly. (I install motion via apt instead of wget).
There are no errors during the installation, but afterwards myip:8765 refuses to connect.
Running systemctl shows that motioneye.service is loaded but shows failed for active and running.
Any idea how I can troubleshoot this further?

starbasessd, 2021-12-01:
What is the messages in 
systemctl status motioneye 
specifically.
What do you get for 
motion --version

nikscha, 2021-12-02:
Motion is version 4.1.1
[image]

starbasessd, 2021-12-02:
Need the complete message for 
sudo systemctl status motioneye 
for example I get:
[image]

nikscha, 2021-12-02:
Ahh okay, my bad.
Here you go:

pi@arges:~ $ sudo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2021-12-01 20:58:39 CET; 17h ago
  Process: 2188 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 2188 (code=exited, status=1/FAILURE)

Dec 01 20:58:38 arges systemd[1]: Started motionEye Server.
Dec 01 20:58:38 arges meyectl[2188]: Traceback (most recent call last):
Dec 01 20:58:38 arges meyectl[2188]:   File "/usr/local/bin/meyectl", line 6, in <module>
Dec 01 20:58:38 arges meyectl[2188]:     from motioneye.meyectl import main
Dec 01 20:58:38 arges meyectl[2188]:   File "/usr/local/lib/python3.7/dist-packages/motioneye/meyectl.py", line 28, in <module>
Dec 01 20:58:38 arges meyectl[2188]:     import settings
Dec 01 20:58:38 arges meyectl[2188]: ModuleNotFoundError: No module named 'settings'
Dec 01 20:58:39 arges systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Dec 01 20:58:39 arges systemd[1]: motioneye.service: Failed with result 'exit-code'.

---------

I checked /usr/local/lib/python3.7/dist-packages/motioneye/ and settings.py IS present. I wrote test.py which imports settings.py and prints "test". Executing test.py fails with the error that settings.py can't import motioneye.py:

pi@arges:/usr/local/lib/python3.7/dist-packages/motioneye $ python test.py
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import settings
  File "/usr/local/lib/python3.7/dist-packages/motioneye/settings.py", line 7, in <module>
    import motioneye
ImportError: No module named motioneye

I checked /usr/local/lib/python3.7/dist-packages/motioneye/ again and motioneye.py **is missing**.
This is the directory:
pi@arges:/usr/local/lib/python3.7/dist-packages/motioneye $ ls
cleanup.py  diskctl.py   __init__.py    meyectl.py     mmalctl.py  motionctl.py  prefs.py     remote.py  sendmail.py  settings.py  smbctl.py  tasks.py     templates  tzctl.py   uploadservices.py  v4l2ctl.py  wifictl.py
config.py   handlers.py  mediafiles.py  mjpgclient.py  monitor.py  powerctl.py   __pycache__  scripts    server.py    shell.py     static     template.py  test.py    update.py  utils.py           webhook.py  wsswitch.py

starbasessd, 2021-12-02:
motionEye is written for python 2.7, not 3.x, and has not been fully updated to support being used in 3.x, yet.
Which set of instructions did you follow, for which OS?

ldispi1, 2021-12-02:
I have the same problem as @nikscha . 
I followed this guide [(Install On Ubuntu (20.04 or Newer)](https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29).
I'm running Ubuntu 20.04.1 LTS x86_64 on a VirtualBox VM.

starbasessd, 2021-12-02:
Checking

starbasessd, 2021-12-02:
What do the commands show:
python --version
python2 --version
python3 --varsion
pip --version
pip2 --version

ldispi1, 2021-12-02:
After running "python --version" I saw that it wasn't installed on my system.
So, I ran "apt install python" and followed the guide once more.
Now everything is working fine.
Thank you very much! @starbasessd

starbasessd, 2021-12-02:
Running VBox 6.1.30
Running VM clone master clean image 
Ubuntu 20.04.3 LTS x86_64   
Resources 4 Cores/4GB/50GB for VM
Updated to today
Installed following 
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Line for line, no copy/paste blocks
All installed correctly. 
motioneye came up on http://<ip_address>:8765
No issues. Last tested date in wiki updated to today.

nikscha, 2021-12-03:
> motionEye is written for python 2.7, not 3.x, and has not been fully updated to support being used in 3.x, yet. Which set of instructions did you follow, for which OS?

I followed the Raspian Buster instructions.

python and python2 are both version 2.7.16, python3 is 3.7.3, pip and pip2 are both 18.1. and located in the python 2.7 directory.

starbasessd, 2021-12-03:
@nikscha 
Need the complete message for
sudo systemctl status motioneye
for example I get:
[image]

nikscha, 2021-12-03:
Fixxed it!
You were right about the python version causing the problem.
Uninstalling motioneye and reinstalling with pip solved it for me. 
Thanks for you patience and your help!

closed 2021-12-03 by nikscha