motioneye-project/motioneye #2243: IMPORTANT INFO: Bullseye CSI Camera Support & Raaspberry Pi

starbasessd, 2021-12-05:
It would appear that at least part of the PiCamera support has been ported to Bullseye, according to this story:
https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/
Scroll down a bit to the particular section on how to grab the 'update'. 
No word yet on the Python functionality...

jedtrently, 2022-01-17:
> It would appear that at least part of the PiCamera support has been ported to Bullseye, according to this story: https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/ [...]

I enabled legacy camera support recommended in article and my VNC now reads 'cannot display desktop' was working fine before I enabled it, now disabling it doesn't change it?

Any ideas?

Cheers

starbasessd, 2022-01-17:
It's not fully compatible. If you are using CSI cameras you are going to need to use Buster Legacy or Buster from May 10 2021. Whatever the broke, left out, adjusted still breaks a lot of stuff including THEIR classroom lessons on the CSI Cameras that they have posted on their website, and encouraged others to write prior to their Bullseye release.

JamBalaya56562, 2026-10-30:
Thanks for the note. The legacy camera stack it refers to is no longer available since Raspberry Pi OS Bookworm, so this no longer applies. Support for CSI cameras with libcamera is tracked in #2683 and #2812, and the wiki has the current steps under Troubleshooting → libcamera camera stack. Closing this as outdated.

closed 2026-10-30 by JamBalaya56562